kwindrem/GuiMods #93: Suggestion/wish if no DC system there don’t show this….

testpaul999, 2022-12-11:
Hi,
My big respect for this project very helpful.
I have a little wish is it possible build a screen like this
[image]
I don’t use a dc system and it is not active at my Venus OS.

regards

kwindrem, 2022-12-11:
The DC System tile showing up when "Has DC system" is turned off is a bug in my code. I'll fix that.

I'm not sure why the flow line to AC Loads is not showing up. Tell me a bit more about your system. Specifically where your AC loads are connected: to the AC input? to AC 2 out?

Also, I see L2 and L3 are showing zero. Is that correct for your system? Is that the 3-phase grid meter on a single phase system bug I saw reported on the Victron forum?

testpaul999, 2022-12-12:
Hi,
and thx for feedback.
I have a zero point system with only one summary smartmeter. (Tasmota via IR readout)
So that only show L1 as summ of all phases.
Only use AC-In.
„that the 3-phase grid meter on a single phase system bug I saw reported on the Victron forum?“ 
Think yes, all other phases commend out at my script but was shown.
Regards

kwindrem, 2022-12-14:
I can't find the reason the flow line between Multi and AC output is not present.
Please confirm "Inactive Tiles on Flow Overview" in Settings / Display & Language / Gui Mods  is not turned on.

(I did find the bug in showing/hiding the DC System tile.)

testpaul999, 2022-12-15:
If I set
"Inactive Tiles on Flow Overview" = hide
I get this overwiew 
[image]
Don’t like this 

"Inactive Tiles on Flow Overview" = full
I get this overview 
[image]

at system Setup 
[image]
no DC System active.

kwindrem, 2022-12-15:
That explains a lot. The option to show inactive tiles is for those that don't like white space on the page. Some prefer empty tiles over the empty space.

But the problem is in the AC output tile. For whatever reason, it's not showing even though the tile is in fact showing values. I need your help figuring out why.

With "Inactive Tiles on Flow Overview" set to "Hide":

What does "Victron stock" overview show?

What does the "GuiMods AC Coupled" overview show?

testpaul999, 2022-12-15:
With "Inactive Tiles on Flow Overview" set to "Hide":
[image]
What does "Victron stock" overview show:
[image]
What does the "GuiMods AC Coupled" overview show:
[image]

hope this helps

kwindrem, 2022-12-15:
Thanks. This gives me something to go on. The issue is the stock "simple" (OverviewHub) flow page isn't used for systems like yours with loads on the AC input. Instead, it uses a different flow overview page: OverviewGridParallel as shown in your second picture. OverviewHub doesn't support loads on the AC input and I based my GuiModsSimple flow page on OverviewHub. I'm pretty sure I can make the necessary changes now that I know what's going on.

testpaul999, 2022-12-15:
🤩

kwindrem, 2022-12-16:
Give GuiMods v7.5 a try. I think it's fixed but do not have a system with loads only on the AC input.

You should set "Inactive Tiles on Flow Overview" to Hide, then only the active tiles will be displayed.

Note that v7.4 was more or less correct in that the AC Output tile was not displayed because there are no loads on the output of the Multi/Quattro. The stock code senses this and changes to the Grid Parallel flow page. My AC Coupled page is actually a better choice as it shows the loads where they actually are. To be completely correct, the AC Out Loads tile should not be shown but I don't think I have a way to hide it. There is a setting: Has AC Out System but this is only accessible if you have an ESS system. In your case you do and could turn this off.

In any case, v7.5 GuiMods Simple should now do what you expected, and the DC System tile should not appear. (Actually the DC System appeared in your system with v7.4 because you had "Inactive Tiles on Flow Overview" set to Full, but I did change the logic any way.)

kwindrem, 2022-12-16:
My fix was incomplete. try v7.6

testpaul999, 2022-12-16:
Well done 🤩
[image]
This are what I won’t to have.
Great job.
Thx

closed 2022-12-16 by kwindrem