kwindrem/GuiMods #95: Active packages (tap to edit) unresponsive

Justice12t, 2022-12-23:
I didn't notice it until v7.8 but when I try to click on any of my active packages nothing happens.  This makes it so I'm not able to update to the latest V.  I've tried several approaches (PC, mobile device, ect) all with the same result.  I also noticed on the package manager page, "action to finish install/uninstall" but that doesn't let me click it either or tell me anymore information.  Honestly, I'm not sure when that showed up.  Currently running OS 3.14.

kwindrem, 2022-12-23:
I see this also. This would be SetupHelper, not GuiMods.

I'll look into it.

kwindrem, 2022-12-23:
I just pushed SetupHelper v4.30 that should fix the issue. If you have automatic download/update turned on your system should update automatically. If not, you can turn it on for a one-time update.

Justice12t, 2022-12-23:
downloaded and appears to be resolved.  Thanks!

closed 2022-12-23 by Justice12t