kwindrem/GuiMods #97: Flow overview modification - GuiMods DC Coupled

jrolleyecora, 2022-12-23:
Hi,
First of all, congratulations for the great job you are doing.
Would it be possible to make such a change? It would be something similar to GuiMods DC Coupled but without the Wind Generation and Fuel Cell box. What modifications should I make? What files should I modify? I would also like to see the temperature of the battery as can be seen in the image

[image]

kwindrem, 2022-12-23:
That page is in the file OverviewFlowComplex.qml. You can delete the items you don't want and expand the size of the tiles you want bigger.

Your AC Loads tile is similar to what I provide in the GuiMods Simple page so you could lift that from Overview HubEnhanced.qml.

Temperatures are displayed as mini-tiles along the bottom if they are included in the dBus system. Adding temperature to the battery tile would be easy but you need to remove something else.

jrolleyecora, 2022-12-24:
Thanks, I'll try it...

closed 2023-03-22 by kwindrem