kwindrem/GuiMods #98: Winter mode

ChillV2L, 2022-12-27:
Hi first of all thanks for all your efforts they have really improved my experience with my victron project incorporating V2L project (vehicle to load) using my victron quattro, 
any way i was wondering if there was a possibility to add "winter mode" to the ESS with the option to select one of the auxiliary inputs (this used to be in the old assistant "self-consumption Hub-2-v3) and i really miss this as part of my home automation setup to boost the charge when my energy is on cheap rate outside the normal off peak periods (octopus intelligent (UK) ) if your interested.
Many thanks again 
Chris

kwindrem, 2022-12-27:
This is beyond what GuiMods could accomplish and I really do not want to get into modifying ESS logic.

closed 2022-12-27 by kwindrem

ChillV2L, 2022-12-27:
Ok thanks anyway for taking the time to reply , and again thanks for mods and improvements you keep making