kwindrem/GuiMods #99: AC Loads shows only L1 at GuiMods simple

testpaul999, 2023-01-01:
Hello,

have running the lates GuiMods with the simple Flow overview.
I have now install a three phase smart meter like EM24 (Elgris)
[image]

This are the overview
[image]

Victron stock will show this
[image]
So You see all phases.

thx for help and a happy new year :-)

kwindrem, 2023-01-01:
Hi Paul.

Where is your smart meter installed (grid connection?)?

Is the Multi a single phase or three phase system?

—
Kevin

testpaul999, 2023-01-01:
Hi,

the smart meter installed at the grid after the EVU smart meter before the whole power distributor.
I have a single phase MultiPlus.
But only voltage and current will fail.
Regards

kwindrem, 2023-01-01:
Thanks. I know what the problem is: For voltage and current, I'm only looking at the Multi/Quattro AC output which in your system is only one phase. systemcalc provides the power numbers so I'll need to do what it does in terms of collecting power for the phases without an inverter.

I suspect this will also be an issue on the details page.

testpaul999, 2023-01-02:
One thing more :-( 
The AC site doesn’t display negative power
[image]

kwindrem, 2023-01-02:
If the PV inverters in this system are on the AC input, then I believe the values shown are correct. systemcalc includes PV in the AC output consumption parameters but not of the PV on AC input. 

Could you check to see what the stock flow shows?

There is limited space on GuiMods Simple. Switching to GuiMods AC Coupled will eliminate details in the tiles but should show PV inverter contribution on both input and output.

testpaul999, 2023-01-02:
I‘ll do so if we have sun tomorrow :-)
„Could you check to see what the stock flow shows?“
Ihave only MPPT charger and no inverter at AC input.

kwindrem, 2023-01-02:
Interesting. Seems like the AC out tile suggests the Multi is on L1. But the AC input tile suggests it's on L2 since the only way to get negative on the AC in is for the Multi to be feeding to the grid.

Where are the AC out loads connected: Multi AC input ro Multi AC output?

testpaul999, 2023-01-03:
All loads are on AC-in I do zeropoint feedin.

kwindrem, 2023-01-09:
Sorry this is taking so long. I'm still trying to unravel the logic in systemcalc for the consumption values that eventually populate the TWO AC load tiles. The logic is very complicated, taking into account loads and PV inverter contributions on both AC inputs even if not active, with/without a Multi (or Quattro) on a specific phase.

Your system is fairly simple as there are no critical loads so only one AC output tile. But I need to accommodate other system configurations as well.

Recent changes to systemcalc have added current calculations (early versions only included power) which helps but still doesn't provide voltage or frequency. 

Still working on it though.

testpaul999, 2023-01-12:
Thx for update 🙂
Don’t hurry it’s only cosmetic.

kwindrem, 2023-03-22:
My previous comment is incorrect and was actually related to another issue, not the one you raised.

These changes are not in v7.17 but I just pushed a beta version  (V7.30~4) which do include the changes. I'd appreciate you testing things out.

Change the branch/tag for GuiMods to "beta" using PackageManager / Active Packages.

This ended up being a rather involved change to systemcalc as other options just got too involved.