kwindrem/GuiMods #102: Possible interference with Wi-Fi connection on Cerbo GX?

aplord, 2023-01-17:
I had the current version of guimods installed on a Cerbo GX, it worked fine for some months, great software thanks. It was connected by Wi-Fi to a Starlink Wi-Fi router. Then some weeks ago the Wi-Fi link to the Starlink would drop unexpectedly. Sometimes it would reconnect but sometimes it would remain disconnected until I connected via bluetooth VictronConnect app and pressed the reconnect button in there. A pain because I am often 100 miles away for weeks at a time.

After trying many things to resolve the problem, in desperation, I finally removed guimods from the Cerbo GX. And the problem immediately ceased. I can't see how guimods could affect the Wi-Fi link, and it might just be a coincidence. But I thought I would mention it here in case you can see any possible link.

Other things I tried unsuccessfully before I removed guimods:

1. Move router and Cerbo nearer/further apart - they are usually 1 metre apart
2. Set a fixed IP address on the Cerbo GX
3. Cold restart Cerbo GX (unplug for > 60 seconds)
4. Factory reset Starlink.
5. I read from Victron of a hardware bug in the Cerbo GX whereby they had put the Wi-Fi antennas next to the HDMI output and they conflicted. They described a fix. I checked mine and it was manufactured after that problem. I pulled the back cover off to confirm.

Thanks if you can help and sorry if this is a spurious report!

kwindrem, 2023-01-17:
I can't see there is any way your WiFi issues could be related to GuiMods.

rspott, 2023-01-19:
Consider this forum post: https://community.victronenergy.com/questions/82877/cerbo-gx-wi-fi-issue-when-having-the-gx-touch-conn.html

and this PDF sent to dealers: https://community.victronenergy.com/storage/attachments/18497-2021-03-cerbo-gx-wi-fi-issues-when-combined-with-g.pdf

I have had a very non-technical friend perform this fix and things were greatly improved.

aplord, 2023-01-19:
> Consider this forum post: https://community.victronenergy.com/questions/82877/cerbo-gx-wi-fi-issue-when-having-the-gx-touch-conn.html

Thanks @rspott, I looked into that. My serial number suggested that my unit was built after the problem had been found. But I pulled the unit apart to check anyway since the symptoms matched. Alas the two Wi-Fi antennae are located in the correct place already 😥