kwindrem/GuiMods #103: Missing data after update

StupidMonkeyKev, 2023-01-18:
[image]

This is my issue, everything was fine till the cerbo updated, then went a bit wack - saw there was updates for setup helper and guimods but I just lost more on the display. I've tried going back to 2.92, no luck, swap to "victron stock" rather than DC coupled and AC input/output boxes come back but still no data for my DC, PV and inverter data/shore power.

Remove GuiMods and display is fine, DC, PV and inverter data/shore power all there on either cerbo firmware.

Any idea?  I've had a right good mess around and a couple of dozen reboots but I'm lost  and I REALLY like the Guimods :) Thanks for doing it for the community. and my current choice is see the temps with GuiMods or power data without

[image]

kwindrem, 2023-01-18:
Can you get to the logs? I'd be interested in the one for systemcalc as that's the only thing that might cause everything to disappear.

tail -50 /data/log/dbus-systemcalc-py/current | tai64nlocal

GuiMods is working fine here on my test RPI and a working system with Cerbo.

kwindrem, 2023-01-18:
You need terminal access either a direct serial connection or via ssh from another computer.

https://www.victronenergy.com/live/ccgx:root_access

Then:

tail -50 /data/log/dbus-systemcalc-py/current | tai64nlocal
tail -50 /data/log/PackageManager/current | tai64nlocal
tail -50 /data/log/SetupHelper | tai64nlocal

zip the results and post here.

PackageManager dumps the last two logs to a USB flash drive when doing a Settings backup, but not systemcalc which is the likely suspect in this case.

You could try uninstalling GuiMods and then SetupHelper, then INSTALLING v2.92 again (not selecting from the firmware backup) and reinstalling SetupHelper and GuiMods and see of the issue persists. If that succeeds, then try INSTALLING v3.00~15 again. Installing replaces all files in the working directories so you have a fresh image. This could resolve the issue if something got corrupted along the way.

kwindrem, 2023-01-19:
Let me know what devices you have connected to your Cerbo and I'll look over the code.

StupidMonkeyKev, 2023-01-19:
I'll try and do the command line stuff tonight, and get at the logs.

Cerbo GX with Multiplus compact 2000-80-30. VE bus connection to Cerbo GX
SOC/Battery via BMV-700 with VE direct connection from remote display to Cerbo
BMV-714??  smartshunt with bluetooth on a non-victron MPPT as a PV DC Power meter - VE direct cable shunt to Cerbo

And a touch 50 screnn

rspott, 2023-01-19:
[image]
[image]

Attached are the results of these commands:
  531  mkdir info
  532  cd info/
  534  cp  /data/log/PackageManager/current ./PackageManager-current
  535  cp /data/log/dbus-systemcalc-py/current ./dbus-systemcalc-py-current
  536  cp /data/log/SetupHelper ./SetupHelper
  541  cd ..
  542  tar cvfz info.tar.gz info/

[info.tar.gz](https://github.com/kwindrem/GuiMods/files/10456805/info.tar.gz)

I hope this helps. Thank you for doing all this work.

rspott, 2023-01-19:
I will need to uninstall so I can visualize the data on my system locally for a bit. 

I am GMT-5 if you need anything. Shoulder surfing etc.

StupidMonkeyKev, 2023-01-19:
[SetupHelper.zip](https://github.com/kwindrem/GuiMods/files/10458381/SetupHelper.zip)

Well it took me way too long to even find the IP for the Cerbo .....  but my old brain caught up and I think I've got that data here for you now.

StupidMonkeyKev, 2023-01-19:
[WITHGUIMODSINST.txt](https://github.com/kwindrem/GuiMods/files/10458493/WITHGUIMODSINST.txt)

Hopefully that helps - all 3 in 1 text file

kwindrem, 2023-01-19:
Thanks. Got what I need and know where the problem is. I'll get a fix out today.

StupidMonkeyKev, 2023-01-19:
[image]

Would appear to have fixed it, thank you

IvanfromVenus, 2023-02-06:
Hello,

After installing v7.13 I get the same or similar error - missing data. Both DC and Solar data are missing.
After installing the guimod, I see don't see anymore the shunt connection under settings/system setup/battery measurements/

[image]
[image]
[image]
[image]
[image]

kwindrem, 2023-02-06:
What Venus OS version?

Can you get to logs? You can do a PackageManager Settings backup which will create a log zip file on removable media that contains all log files. Post that file here and I'll take a look.

IvanfromVenus, 2023-02-06:
Venus OS version is 2.92. the log i will tomorrow

IvanfromVenus, 2023-02-11:
I updated the logs.

kwindrem, 2023-02-12:
The logs showed the problem. Will get a fix out tomorrow.

kwindrem, 2023-02-12:
Should be fixed now in v7.15.

closed 2023-02-12 by kwindrem