kwindrem/GuiMods #104: Shutdown monitor not functioning properly

gainestr, 2023-01-20:
Just tried to run a shutdown.  Even though guimods is showing that shutdown monitor is installed, it apparently is not functioning properly.  
[image]

kwindrem, 2023-01-21:
ShutdownMonitor is actually running. The problem is in the GUI, falsely showing the not running message. I'm working on a fix now.

If you scroll down, you'll find the Shutdown? command and that does work.

kwindrem, 2023-01-23:
Fixed now.

BTW, this issue should have been in the ShutdownMonitor repo as it pertains to code there not in GuiMods.

gainestr, 2023-01-23:
Thank you and remembered for next time.