kwindrem/GuiMods #105: Scaling issue with v7.12?

Justice12t, 2023-01-29:
Hi,

I'm on 3.00 ~14 and just updated to v7.12.  After that I noticed the DC systems tile looks to be scaled incorrectly.  I've attached a picture.  Just wanted to see if anyone else noticed this.  Thanks
[image]

kwindrem, 2023-01-30:
I am not seeing this on my systems and the code I'm using for the Multi tile hadn't changed -- ever.

 The DC System tile is sized relative to the Multi tile which as you can see is stretched horizontally in your screen shot. Unlike other tiles, the Multi is actually an image file.

On closer inspection, other objects are incorrect also. Notice the connection dots are not round.

I'm not sure why this is happening but may have to do with your screen resolution. The GUI code is based on the original CCGX screen resolution and scales everything back to that and it's possible the scaling of an image is not correct if the screen doesn't have the expected aspect ratio. But I'm not sure why this would have changed in v3.00~14

I'd be interested to know what the "Victron Stock" flow page looks like. You can get to this from a menu item in the GuiMods menu or by uninstalling GuiMods completely.

I'll try to fix the Multi tile width in a future release of GuiMods.

Justice12t, 2023-01-30:
Yea, this is the first time I noticed anything different.  Here's a shot with it uninstalled.  I'll try to investigate a little further tomorrow and see if something else could be causing this and it was just coincidence with the update.  Thanks for the reply.
[image]

kwindrem, 2023-01-30:
Thanks for the prompt reply. 

If you have a physical display connected, what is it's resolution? That could be throwing things off and there might be differences between v2.9x and v3.00~15. If that's the case, this should be reported in the beta test thread in the Victron forum. Use the stock screen shots since Victron wants a "stock" install.

also what platform? (Cerbo, CCGX, Raspberry, etc).

Justice12t, 2023-01-30:
I do, I'll have to double check tomorrow as I don't remember off the top of my head.  I'll try re-seating the cable tomorrow and see if the settings on the display changed for some reasons.  I'm on v3.00~14, but will upgrade to ~15 if the cable doesn't fix it and report it there.  Thanks again for the help.

Justice12t, 2023-01-31:
1024x768 is what the screen said.  I re-seated the cables and still nothing.  It's gotta be something with my Cerbo as I've tried/uninstalled everything else.  Thanks for taking a quick look.

closed 2023-02-01 by Justice12t