kwindrem/GuiMods #106: Battery discharge / charge reversed in details page

teefixx, 2023-01-31:
Hi

In the overview the battery gauge shows "charge" even if I discharge the battery. The values seem to be ok with a - sign in front.

In the details page for the battery shows "charge" as well. The values there have the wrong sign. 

I'm using GuiMods on a raspberry Pi 4, VenusOS 3.00-14, GuiMods 7.12

Thank you for the great UI improvement!

[image]

[image]

kwindrem, 2023-01-31:
Sorry, I'm not understanding the problem. In both screen shots the gauge is showing discharge (towards the "D").

In the overview, I see a negative number which is consistent with the battery discharging.

In the detail I see "supplying" which indicates the battery is supplying power to the rest of the system. This would say "consuming" if the battery was taking energy from the rest of the system (aka "charging") I display only positive numbers here because of the wording right next to the power and current numbers. It would be incorrect to say the battery is supplying -910 watts. That would mean it was actually consuming.

I kept the positive and negative power and currents as this is the way Victron did it initially and didn't want to make a change.

Other gauges show "S" for supplying and "C" for consuming while the battery shows "D" for discharging and "C" for charging. I did this because charging and discharging are common terms when referring to battery state. If you have better terminology I'd consider a change.

teefixx, 2023-01-31:
Hi kwindrem

Your're right with the gauge, sorry for my mistake and thank you for the explanation of the + and - signs.

One thing tough I forgot to mention: The PV inverter does not show the gauge even if I defined it int the settings.
Is there anything I can do about it?

thank you

kwindrem, 2023-01-31:
Looks like you found a bug. Thanks. There are three locations for PV inverters: grid input, generator inputi and output (load) of the Multi.

Victron selects one of these locations for display in the stock simple flow and I copied that code. If there is a PV inverter on the output the PV on input is not included in the total power.

The other issue is that regardless of the PV inverter location, I use the PV On AC output max power for the gauge.

I think this tile should show the combined power of all PV inverters regardless of their location.

Until I fix this, you may be able to get the gauge to work by setting both PV on input and PV on output max power to the same value.

teefixx, 2023-01-31:
Hi
not a big deal.

FYI: setting the same value at PV inverter at output and input does not help either...

closed 2023-03-03 by teefixx