kwindrem/GuiMods #108: Battery Temperature displaying twice

Fraser86, 2023-02-03:
When I have dvcc activated I am seeing the battery temperature twice. Is there a way to turn this on and off
[image]

kwindrem, 2023-02-03:
That depends on the source of the temperature information. I look for a battery monitor separately from other available temperature devices and do have a user preference to turn off the battery monitor temperature. That might help.

Fraser86, 2023-02-03:
I Like to see the temperature of my lithium battery, due to it being positioned in a boat and temperature is a factor when charging. 
I Connected my battery and the temp info displays automatically. 
I then switched dvcc on to use bms as battery monitor and it displays the temperature twice. 
If i go in display then Gui Mods, and try to switch off display battery, both are removed. 
Im not sure not sure how to turn one of these off.

kwindrem, 2023-02-03:
What temperatures are showing in the device list?

Fraser86, 2023-02-04:
Only the CPU of Pi in device list

kwindrem, 2023-02-04:
I have no idea what is going on. Is it possible DVCC is doubling up what's shown as BMS?

Fraser86, 2023-02-04:
I have SerialBattery(jkbms) runnning. Inside that is the temp of the battery. 

Imside the dvcc menu 
STS - shared temperature sense

I have selelected the serialbattery(jkbms) to insure proper changing temperatures. 

There are 4 options 
Serialbms
Multiplus
Automatic 
No sensor. 

Tried all of these. Same result, temp shows up twice

kwindrem, 2023-02-04:
The only thing I can think of is that somehow the combination of DVCC and your BMS are creating two battery dBus services. What shows up in the device list? DVCC on and off?

Fraser86, 2023-02-04:
If i switch dvcc off i see this message
Make sure you also reset ve.bus system, im not sure how you do that

However i still see two battery temperatures

kwindrem, 2023-02-04:
ve.bus is your Multi/Quattro

Fraser86, 2023-02-04:
So reset multi? 
I could try that. 
Maybe that removes second instance of temp appearing. 

I guess it has something to do with shareing the temperature within the dvcc and that then appears as a temperature from two devices. 

I have little understanding of dbus and what it does so not sure how and why it does this

Fraser86, 2023-02-04:
I am trying to learn but not a coder

kwindrem, 2023-02-04:
dbus is the mechanism Venus OS uses to pass values between sections of code.

I would look at the Devices list and see if there are two battery monitors shown. If so my code that displays temperature will not be able to know that one should be shown and one should not. I actually don't see any other way that I could be showing two temperatures.

Fraser86, 2023-02-04:
There is only one temp sensor and that is my bms.. 
In the device list there are the two battery monitors, 
I.e my bms and my bmv712. 

The bmv does not have a temperature sensor. 
The only temperature displaying in device list is the cpu of the pi.

iLeeeZi, 2023-02-05:
The duplicate temperature is probably coming from multiplus. When STS is enabled, multiplus publishes rounded value of the BMS temperature under multiplus dbus objects Dc/0/Temperature and BatterySense/Temperature. I just disabled STS for now to get rid of the dublicate temperature from main screen but it would be nice to automatically hide it if possible.

Justice12t, 2023-02-05:
Yea being able to toggle which battery temp sensor to display would be great.  I run into a similar problem as above with my BMV712 and VE Smart dognle.  The battery temp shows twice when the inverter is on.  If the inverter is off, I just reboot my Cerbo and it removes the smart dongle entry.

StupidMonkeyKev, 2023-02-06:
I did have this too - drop one of my Ruuvi tags and show the battery twice (also a temp feed from multiplus) but then it comes back. Not had the issue for a week or so - was shortly after the last update. I assumed it was my Ruuvi dropping out rather than a GuiMods issue.

andi79h, 2023-06-24:
I have a related problem: I got three victron sensors to keep an eye on the three batteries and in addition, I got 4 ruuvis for cooler, freezer and two cabin temps. I set up the three sensors to type battery and since battery temps should not be displayed on the overview, I'm wondering why they are there.

Now I already logged into the cerbo and tried a bit of cat and mouse with those qml files. I was able to get rid of ALL temps in this view, yet not specific ones and its a bit frustrating without a debugger. By the way, is there a way to reload the screen without rebooting? :)

[image]
[image]
[image]

Cheers mate

kwindrem, 2023-06-24:
Some people requested showing battery temperatures so I made extra effort to add them to the list.

There is an option in Display & Language / Gui Mods to show battery temperatures in the flow overview, or not. This only affects the flow overview. Battery temps will always appear in the Mobile Overview and the Tanks, Temps, Digital Inputs overview.

The code that adds temps to the list is in addService (). You can remove the battery temps everywhere by removing the case DBusService.DBUS_SERVICE_BATTERY: section. There will be an addService () in each of the overview pages that displays temps.

You can restart the GUI with:

killall gui

or svc -t /service/gui

andi79h, 2023-06-24:
Well this overview I've posted is the flow overview, isnt it? The blue temp rectangles (without humidity %) are always shown, even if I disable "Show battery temp in flow overview". Maybe because it is a victron temp device and not some sort of internal sensor, yet I need a way to get rid of them in this view :)

[image]

What I thought the toggle works was it checks for all temperatur objects if they are of type "battery" and then ignores them. You know, like just passing those printed yellow and green. 
I already was in the flow qml and played a bit with the switch (added an extra break) yet was not able to distinguish the objects in this switch by type as there is just a name used. Another idea was to ignore them by name e.g. "12V AKKU", but that didnt work either. Do they have some internal name used in addService? 

And what is the best way to look into those objects? Is there something like console.log(object)?

Cheers mate

kwindrem, 2023-06-24:
I just tried it and it worked but you need to refresh the page by switching to the menu and back again.

This only works on battery services. Any other temperature service would not be hidden. It appears that the 12V AKKU temp is NOT from a battery service and would not be filtered out. It would be added under case DBusService.DBUS_SERVICE_TEMPERATURE_SENSOR

It is a bit tricky to get dbus parameters from the service object. The hack I used to check to see if there is a temperature could be expanded to get name. The parameter is /CustomName.

andi79h, 2023-06-24:
Yeah, that's what I thought, I already was in the case DBusService.DBUS_SERVICE_TEMPERATURE_SENSOR part.

So like 
`property string tempServiceCustomName: ""`
`property VBusItem temperatureItemName: VBusItem { bind: Utils.path(tempServiceCustomName, "/Dc/0/Temperature/CustomName") }`

to get its name from the display and then throw in a break when name starts with "12" or "48"

kwindrem, 2023-06-24:
Yes, that's the idea