kwindrem/GuiMods #109: current limit always set back to initial value

teefixx, 2023-02-04:
Hi
When I set "current limit" to eg 10A everything looks fine until I press "Accept New". It then switches to the start page and when I go back into the sub screen "AC Input Detail" nothing changed. It's always back to the original value.

[image]

[image]

kwindrem, 2023-02-04:
The Multi/Quattro must be programmed to allow input current limit changes. In addition a digital multi control or certain BMS will prevent current limit changes. I am not checking for these. The stock Victron Mobile Overview does check these items. I guess I should add these.

It also appears I have a bug in displaying the graph in the detail page.

teefixx, 2023-02-04:
Hi
Thank you for your reply. I know thought I know the Multis quite well, but never knew that they can be programmed to allow input current limit changes... 
Can you give me a hint please?
thank you

kwindrem, 2023-02-04:
It is in the General configuration and the parameter is "Overruled by remote". This needs to be on in order to allow input current limit changes from the GX device or Digital Multi Control.

kwindrem, 2023-02-04:
I've changed the AC details page to show "current limit is not adjustable". The buttons to make changes are also hidden.

closed 2023-02-04 by kwindrem