kwindrem/GuiMods #113: GuiMods file set incomplete

StupidMonkeyKev, 2023-02-06:
Updated and rebooted after Setup helper update.
But setuphelper/package editor reports "GuiMods file set incomplete" after downloading 7.13 so cant install.

Cheers again for keeping these mods improving all the time :)

kwindrem, 2023-02-06:
What Venus OS version are you running?

There are two causes for this error:

1) a new Venus OS version was released that is not yet supported by the package

2) A file GuiMods replaces was modified from the one Victron supplied with the release.

In the latter case, the easiest way to recover is to reinstall Venus OS. You can do this by switching to the backup firmware then doing a new on-line install.

kwindrem, 2023-02-06:
I see that v2.93~2 needs a new file set. It'll be up shortly.

closed 2023-02-06 by kwindrem

StupidMonkeyKev, 2023-02-06:
Many thanks - saw it auto updated was working before i saw emails on the
fix.  Yes, was on 2.93~2