kwindrem/GuiMods #116: Multiplus Detailed Info

kittyorianda, 2023-02-08:
Would it be possible that instead of the time on the Multiplus 2 (Inverter) the ampere consumption would be displayed? 

The time I do not see now as so important on the screen.
[image]

kwindrem, 2023-02-09:
Amp-hours consumed really belongs in the battery tile. As you can see, it's full with what's there.

If you have the ability to edit the qml code, anything is possible.

kittyorianda, 2023-02-09:
It is not intended to display the amperes consumed. That really belongs to the battery, I agree. My Multiplus converts the 12V current to 230V and I want to know how many amps this inverter is consuming and I want it to show that on its tile. On details page it shows me that with "DC Power / Current consuming".

kwindrem, 2023-02-27:
I'll admit that the lack of information on the Inverter tile is inconsistent with the rest of the system as displayed on this page. Singing out one parameter of interest is likely to differ with each individual: you want DC current, someone else may want AC power.
The gauge displayed in the inverter tile does provide a relative indication of the AC power consumed or supplied by the inverter, just not a numerical value. I see a graphical relative indication as more important to a numerical value since one needs to read the value the mentally compare it to a known (or possibly unknown) maximum.
In your simple system the DC power and current will be close to the AC Output power as you can see from the Inverter detail you provided.
I am reluctant to make any changes in this area at this time.

closed 2023-02-27 by kwindrem