kwindrem/GuiMods #119: GuiMods v7.16 incompatible with Venus OS v3.00~17 ?

sergios100, 2023-02-17:
Hi!
I just updated to Venus OS v3.00~17 on my raspberry PI 4 and it seems that GuiMods v7.16 is not working anymore.
[image]

Is this a known issue or am I doing someting wrong?

Thanks a lot!

Cheers!

kwindrem, 2023-02-17:
v7.16 should support v3.00~17 so I'm not sure what is going on. It's working here.

Use PackageManager to do a Settings backup to removable media and post the logs.zip file here and I'll take a look.

sergios100, 2023-02-17:
I will do that!
Thanks!

sergios100, 2023-02-17:
Please find the requested Zip-File here:
[logs.zip](https://github.com/kwindrem/GuiMods/files/10772398/logs.zip)

Cheers! Sergio

kwindrem, 2023-02-18:
The install is failing because compatible versions for two files can't be found. I've checked and the v3.00~17 file set includes files that ARE a match for those files in the v3.00~17 release. This usually means something modified the files before the GuiMods setup script runs. The easiest way to recover is to revert to a previous version firmware using Stored firmware, then reinstall the v3.00~17 using online updates. If that fails, try redownloading GuiMods.

gainestr, 2023-02-18:
You may want to enable “Auto install packages” for this update.  Worked fine for me.

sergios100, 2023-02-18:
Hi everyone,

got it to run finally! don't know how exactly, but in the process I **removed the GuiMod package**, "**Initalized the Package Manager**", **enabled "Auto install packages"** set **"Automatic GitHub downloads"** to **"on"**.
After a few seconds the Remote Console went offline, and when it came back, GuiMods and all the other Packages are up and running as before. I did not have to revert to a previous firmware in the process.

So thanks a lot for your support! The Issue is solved for me!

Cheers Sergio

closed 2023-02-18 by sergios100