kwindrem/GuiMods #120: Cannot install on Venus OS v2.93

morphias2004, 2023-02-19:
Package Manager says it's incompatible with V2.93, so doesn't give the Install option.

kwindrem, 2023-02-20:
GuiMods v7.16 installs fine here on Venus OS v2.93. v2.93 support was added to GuiMods as of v7.14.

Try downloading GuiMods again and see if that clears the incompatibility.

Next step would be to switch to the previous firmware in the Stored backup area then do a fresh online update to v2.93.

morphias2004, 2023-02-20:
> GuiMods v7.16 installs fine here on Venus OS v2.93. v2.93 support was added to GuiMods as of v7.14.
> 
> Try downloading GuiMods again and see if that clears the incompatibility.
> 
> Next step would be to switch to the previous firmware in the Stored backup area then do a fresh online update to v2.93.

Issue is now resolved.

For clarification, this occurred on a Victron Cerbo GX following an update from v2.92 to v2.93.

Following a reboot of the Cerbo GX, doing a download again in Package Manager and then rebooting again, GUIMods was automatically reinstalled after the reboot.

closed 2023-02-20 by morphias2004