kwindrem/GuiMods #122: Option Dc box to replace the AC

Next9999, 2023-02-20:
Hello

Could we have an option to replace the AC box for DC...alternator

i know we could used the other screen layout but i lose the detail of my solar

[image]

kwindrem, 2023-03-01:
I have implemented this. Please check out v7.17~1.
It can be accessed by switching the GitHub branch/tag to "beta".
I display the alternator on the Generator Overview and on the GuiMods Simple Overview
There's a new GuiMods option to enable it. Consult the ReadMe for more details.

Please let me know if this works for you.

kwindrem, 2023-03-07:
I would like to release these changes. Next999, please test the changes on the beta branch ASAP and report back here as soon as possible. Thanks.

kwindrem, 2023-03-10:
I have released these changes as v7.17. The beta branch as been deleted. Please change GuiMod's branch/tag back to "latest"