kwindrem/GuiMods #123: Data Usage

lfhoward, 2023-02-22:
Hi Kevin,
I have a Raspberry Pi running Venus OS in my camping trailer. It is connected to a wifi network from a cellular hotspot running on another Raspberry Pi. My data plan shows that since installing GuiMods, the data usage has spiked from 5-7 MB per day up to 16-20 MB per day. I have a simple 500 MB per month data plan for VRM monitoring, so this increase puts me at or above my maximum data limit. Automatic downloads and automatic installation options are OFF. VRM monitoring extended to every 10 minutes on the Venus OS side barely makes a dent, so the additional data is definitely GuiMods. How do I keep the data usage down, and what aspect of GuiMods is using 15+ MB per day in web communications? Thanks in advance,
Lauren

kwindrem, 2023-02-22:
PackageManager queries GitHub once per minute for the version number of one package. This happens whether downloads are on or off.

The data transferred is tiny (less than 20 bytes of payload). I would not have expected this to contribute significantly to internet data consumption. According to your measurements, would be using around 7Kb per check. So clearly, I need to reduce the frequency of those checks.

I think I'm going to disable GitHub version checks if downloads are off. This will eliminate all SetupHelper's internet communications.

lfhoward, 2023-02-22:
Thank you for being so responsive! I will look for a SetupHelper update soon and I can report back in a few days about the difference in data consumption. 

I should also mention that my system is running the ShutdownMonitor and the RpiGpioSetup as well, in case there is any data usage associated with those.

kwindrem, 2023-02-23:
SetupHelper v4.34 includes code to disable the GitHub version checks when automatic downloads are turned off.

I expire version numbers after 20 minutes so people don't assume there aren't any updates available.

In the package edit menu, the version for the current package will refresh that GitHub version once. if auto downloads are off so you can see the current version. I'm hoping this will not cause too much network traffic for you. It won't create any traffic if you don't go into the package edit menu so ...

closed 2023-02-23 by kwindrem

kwindrem, 2023-02-23:
If you leave automatic downloads off, you can always manually update your packages from the package edit menu.

I could add a preference to check for automatic downloads less often, say once per day rather than once per minute if this isn't sufficient for you.

lfhoward, 2023-02-24:
Kevin, your modification of the code worked!

[image]

You can see that I downloaded and installed the modified version of SetupHelper at 18:00 GMT today, and the subsequent data usage is back down. Mission accomplished. I can now keep my inexpensive data plan AND keep GuiMods installed. Best of both. Thank you again for your help and have a great weekend. 

Lauren

kwindrem, 2023-02-24:
Great. Glad I could help you out.

kwindrem, 2023-02-25:
ifhoward, I'm considering some additional changes to the Git Hub version refreshes:

I am slowing the background refreshes to one package every 10 minutes rather than every 1 minute.
This would reduce SetupHelper's data usage from 15 Mb/day to 1.5 Mb/day.

I would refresh all package Git Hub versions when entering the Active packages menu even if auto downloads are turned off.

I would still disable refreshes if auto downloads is turned off. I would also refresh the Git Hub version in the Package edit menu.

From your initial reports, I calculate each Git Hub version refresh consumes about 10 Kb of bandwidth. So every time you enter the Package versions menu there would be 10 Kb/package of data usage. Then each change in package in the Package edit menu would be another 10 Kb. These would occur even if auto downloads is turned off.

GuiMods and SetupHelper are about 1.2 MB each so that data would also count against your monthly usage. If they download. With auto downloads on you wouldn't know when that happens.

Let me know if you think the background refresh rate of one package every 10 minutes falls within your data usage or if you would still run with auto downloads disabled.

lfhoward, 2023-02-28:
Hi Kevin, I think checking for new versions once every 10 mins would be fine, and would only take 45 MB a month rather than 450 MB like before. I like having the option to keep automatic downloads disabled for no web traffic. I probably wouldn’t worry about doing that unless I needed the MB for something else in a given month, like downloading the newest Victron firmware. Thanks for asking my opinion on this. I don’t know if I’m the only person in the world with a 500 MB $11 data plan anymore, but it’s perfect to monitor my trailer’s battery and solar system remotely. I turned a second Raspberry Pi into a cellular hotspot using equipment and data from Sixfab.com. It works great on both T-Mobile & AT&T networks, which is an advantage when traveling.

kwindrem, 2023-02-28:
Thanks ifhoward. I've implemented these changes and v4.38 is now available with them.

Entering the Active packages menu now refreshes all versions IF auto downloads is disabled

Entering the Package edit menu or changing to a different package refreshes one GitHub version IF auto downloads is disabled.

These changes keep what's shown in the menus up to date at the expense of some additional bandwidth.

I'm not sure if others have low monthly data usage but I can see how many would want to save some money for specific low bandwidth things like a VRM connection.

lfhoward, 2023-03-01:
Awesome. Thanks for creating such useful software for the Victron Venus OS & Raspberry Pi, and then for working with us users to make it even better. Looking forward to implementing some of your other software soon, like the one for adding a touchscreen to my Venus Pi system. Cheers!