kwindrem/GuiMods #124: How do to show smartshunt auxiliary input on flow?

ijpitts, 2023-02-24:
I would like to have the auxiliary input of my smartshunt connected to my start battery shown on one of the flows. Is this currently possible? 
[image]
[image]
[image]

kwindrem, 2023-02-24:
While it appears that space is available in the battery tile, it is in fact full, so no space to display the aux battery voltage of the shunt.

closed 2023-02-24 by kwindrem

ijpitts, 2023-02-25:
Any chance we could get a page that just shows all the batteries? I will eventually have house batteries, thruster batteries, generator starting battery, port engine starting batteries, and starboard engine starting batteries. Ian Pitts602-312-4773

kwindrem, 2023-02-26:
Are you suggesting a new page you get to by swiping on the current page?

ijpitts, 2023-02-26:
Yes, one dedicated to only battery levels Ian Pitts602-312-4773