kwindrem/GuiMods #125: Wrong flow direction in AC coupled display

chaerle, 2023-02-25:
On the attached picture, the  flow direction is wrong. The display shows current flowing out if "AC  in loads" instead of into.  The victron standard  display is correct.
[image]

kwindrem, 2023-02-27:
Just so I make the correct changes, please verify:

The segment from the inverter AC input to the "input bus" is based on a parameter from the inverter itself so I'm guessing this one is correct.

Then I believe the segment between the inverter input line to the PV on Input line is backwards based on your drawing.

The segment from the AC In Loads to the PV on Input line is also backwards.

All correct?

chaerle, 2023-02-28:
Yes

chaerle, 2023-02-28:
Also there is no PV on output. Can this be hidden?

kwindrem, 2023-02-28:
Thanks for the confirmation. I found the issue with PV on Output and will fix that also.

kwindrem, 2023-03-01:
I have a beta version that includes a fix for the Loads on Output flow direction and correctly hiding the PV on Output

In PackageManager / Active packages / GitHub branch or tag, change to "beta" to download/install this version.

Please have a look and make sure It's working correctly.

You can stay on the beta branch to continue to receive updates but as soon as the beta test is over, you need to switch back to "latest" to get future updates.

chaerle, 2023-03-01:
Problem fixed.

closed 2023-03-01 by chaerle