kwindrem/GuiMods #128: Time in overview not centered

The2Earthlings, 2023-03-04:
I have a clean install with lautest Venus OS with guimods 7.16 on my rpi and the overview shows the time not centered.
Can you pls change this?

thx Norbert 
[image]

kwindrem, 2023-03-04:
The time is off center to allow space the DC connection from the Multi.
[image]

closed 2023-03-04 by kwindrem

The2Earthlings, 2023-03-05:
@kwindrem but if no connection available to a Multi in System (e.g. No Multi in System) it should be centered.

kwindrem, 2023-03-05:
In reality, the Multi tile should not be shown at all since there really isn't one.

The2Earthlings, 2023-03-05:
So the displayed multi is a bug? Does it make sense then to remove it the multi in guimods view? (And display only the time)?

kwindrem, 2023-03-05:
Technically, components that do not exist don't appear on the flow page. Victorn's stock flow page always shows the inverter and battery even if they do not exist. I didn't change that. The stock flow also shows the inverter connections so the DC one would always be present. If no components exist, this would create a blank page which isn't a good thing as it looks like a GUI crash.

With such a minimal system, you might look at the GuiMods Simple flow page.

I'll look to see what changes make sense.

The2Earthlings, 2023-03-07:
Thx a lot. Will you still display the time if the invertier tile is hidden? Maybe a bit bigger and centered and a bit more on top.

kwindrem, 2023-03-07:
I'm not hiding the inverter tile

The2Earthlings, 2023-03-07:
Oh yes sorry.