kwindrem/GuiMods #129: inverter detail "total power" shows L1 power instead of sum of L1 L2 L3

teefixx, 2023-03-05:
When I display the inverter details page, the value for "total power" shows only L+ power.

The gauge shows only L1 and L3 (at least it looks like)

[image]

kwindrem, 2023-03-05:
I've looked at the code and don't see anything obvious. Tell me more about your system like where you have PV inverters and loads and if you have a grid meter. Of special interest is if you have inverters on all three phases.

Most curious that our output currents are negative. And that the inverter powers are negative yet DC power indicates the inverter is supplying power.

teefixx, 2023-03-05:
Hi

VenusOS 300-21 on RPi 3b
3x Multiplus 48/5000, each on one phase
1x SE12.5k inverter on ACout
1x MPPT150/60
1x Gavazzi EM540 grid meter

all loads on ACout, no loads before ACin

In the situation illustrated on the pic the inverter was charging the battery and supplying ACout loads.

kwindrem, 2023-03-06:
I think I found both problems. I'm currently beta testing some other changes in GuiMods. You can check the fix by changing the GitHub branch/tag for GuiMods to "beta". This is in Settings / PackageManager / Active packages, then tap on GuiMods.

teefixx, 2023-03-06:
Power is now OK, it sums up the phases.
Gauge still shows only L1 and L3

kwindrem, 2023-03-06:
OK, I think I got the gauge working now. v7.17~6

kwindrem, 2023-03-10:
I have released these changes as v7.17. The beta branch as been deleted. Please change GuiMod's branch/tag back to "latest"

closed 2023-03-10 by kwindrem