kwindrem/GuiMods #130: Guimods - Venus OS Watt display stops

Gexle, 2023-03-08:
Hello Kevin

As discussed today's log file...

Thank you
[logs.zip](https://github.com/kwindrem/GuiMods/files/10918070/logs.zip)

kwindrem, 2023-03-08:
I see indications that a PV inverter is coming and going.

From systemcalc:

@400000006408334005c8d3cc INFO:dbusmonitor:       com.victronenergy.pvinverter.http_40 was skipped because it has no device instance
@40000000640833ac08c44534 INFO:dbusmonitor:Found: com.victronenergy.pvinverter.http_40, scanning and storing items

and from the gui:

@400000006408303a15e9b014 Error value: QDBusError("org.freedesktop.DBus.Error.NoReply", "Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.") "com.victronenergy.pvinverter.http_40" "/CustomName" 
@400000006408303a15eb3e84 Error value: QDBusError("org.freedesktop.DBus.Error.NoReply", "Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.") "com.victronenergy.pvinverter.http_40" "/ProductName" 

GuiMods attempts to get /CustomName but not /ProductName so I'm not sure where this is coming from. It's possible the request is causing the PV inverter to drop the connection. This could be causing a hang somewhere while the connection is down. Just a guess.

What PV inverters do you have in your system?

Gexle, 2023-03-08:
Hello Kevin

I saw that too!

But I don't have a PV inverter installed in my ESS system!

kwindrem, 2023-03-08:
There is something in your system showing up as  PV inverter. Device discovery scans all dbus services and "com.victronenergy.pvinverter..." is being found. It looks like it is on a network connection (...http...)

What components do you have in your system?

How are you connected to the internet for VRM logging?

Anything else on this network that might be triggering this discovery?

I would uninstall GuiMods, reboot and check the logs for these same messages. The GUI pausing may be in indicator of something else going on in your system.

I would then reinstall GuiMods then start disconnecting components from the system one at a time to see if you can isolate the issue to one component.

Gexle, 2023-03-08:
There is one but as a full feeder system that is connected to the network, i.e. to my home network.

Can I disable this query somehow?

I don't know how that should come about.

Here is a picture of my tools!
[image]

Gexle, 2023-03-08:
Hello Kevin

Some time ago I copied this folder because I wanted to integrate my Kostal Piko.

I just copied the folder and that's it, I didn't install anything or anything else, just entered the data for my Piko and it says http: inside.

https://github.com/schenlap/venus_kostal_pico

If I delete the folder would everything be OK?

As I said, I just copied the folder according to Github and entered my data!

Didn't do anything or anything!

kwindrem, 2023-03-08:
Regarding Kostal Piko: If you did any of the setup steps, you may have a /data/rc.Local file which will start up the Piko monitor. You'd need to remove this. Having the folder in /data won't do anything by itself.

There is no way to prevent Venus OS from attempting to discover devices on the LAN. Your PV inverter may be getting sucked into your Venus system although it is not showing in the VRM device list so it's not completely there. You'll probably need to isolate things to separate networks.

Gexle, 2023-03-08:
Ok, that means I'll delete the Kostal Piko folder, right?

What do I need to delete in the date/ folder?

kwindrem, 2023-03-08:
do NOT delete /data !!!! It contains other things

/data/rc.Local is a file that is executed at system startup. It is often used to start processes or do other things some extension to Venus OS needs to do at system startup. If this file exists, you need to look inside it to make sure it only contains lines relating to the Piko.

`cat /data/rc.Local
`
will display the contents of the file.

If it only contains things relating to Piko then:

`rm /data/rc.Local
`
If not, you need to use an editor like nano to to remove the Piko lines from the file.

Gexle, 2023-03-08:
Ok all understood
I'll take a look later and if I'm not sure I'll post the file here again for you to look at.

Thanks Kevin for your help!

Gexle, 2023-03-08:
Hello Kevin here is my rc.local

is this so ok to Kostal Piko I have not found anything
Shelly 3em I didn't just create a folder either.

Link 
https://www.transfernow.net/dl/20230308Ss6ySXew

Gexle, 2023-03-08:
This Shelly 3em I also not only have the folder, it could not be this one, it is also called PVInverter
[image]
[image]

Soll ich diesen mal löschen?

Und schau der Current.log wird auch geschrieben!
[image]

kwindrem, 2023-03-08:
Piko is NOT included in your rc.Local but there are other things. 

One modifies the GUI startup script but not sure from where that came from. Something placed a start-gui.sh file in /data. The other stuff there installs touch screen calibration stuff and updates the pointer calibration. as well as backlight and display blanking. I suspect it's an alternative to my RpiDisplaySetup package.

The last line runs a shelly smart meter install script with each boot.

The Piko folder is not running anything on your system and can be removed if you choose but it won't change any behavior.

At this point you need to do some troubleshooting. For example, uninstall GuiMods and see if you still see the errors in say the systemcal log file.

Disconnect your Venus PI from the LAN and check for those error messages.

While disconnected from the LAN, reinstall GuiMods and check the logs again.

Not sure you've done this already but you could upgrade/downgrade to a different Venus OS then reinstall the latest using on-line updates. This will remove any modifications most things make to the operating files. That said, SetupHelper will reinstall all of my packages or any third-party packages included in the Active packages list after the firmware update. You can prevent this by uninstalling them BEFORE doing the online update.

kwindrem, 2023-03-08:
This could be your issue. The device instance sent in the ini file is 40 and that same number is attached to the http portion of the service name.

Check the Shelly documentation for the proper way to uninstall as simply removing files could cause other issues. As I mentioned, your rc.local file contains a call to a shelly install script which will need to be removed, possibly manually. But there may also be operating files that need to be removed too. Again, check Shelly documentation. I can't help with this.

Gexle, 2023-03-09:
Hello Kevin

I have now deleted the line with the Shelly 3em RPi in the rc.local and everything is OK so far..no more stoppers and no error entries in the current log.

Thanks for your help

kwindrem, 2023-03-09:
Excellent. Glad it was simple

closed 2023-03-09 by kwindrem