kwindrem/GuiMods #131: Naming tanks from SeeLevel N2K tank monitor

jnm30327, 2023-03-09:
I can rename the tanks where data is from Bluetooth sensors in GuidMods.

I don't see a way to rename tanks where data is from VE.Can connection to SeeLevel N2K monitor. I can choose the material in the tanks. ie. Waste water, black water (sewage), and fresh water. But these are concatenated in the Cerbo display and the VRM widget display with "Tank 0", "Tank 1", and "Tank 2". 

Is there a way to get rid of the "tank" names? "Shorten tank names" helps on  temps, digital inputs display but not in device names,  tank display, or VRM widgets.

Thanks!

kwindrem, 2023-03-09:
Sorry, no. Victron made the decision not to provide local names for any N2K tank sensors. They see this as the responsibility of the sensor manufacturer. Victron fails to see that with things like SeeLevel, we are stuck with fixed names that may not make sense.

You should post an issue on the Victron Community forum. I've seen the voice of many changing Victron's mind, like the addition of RV-C tank sensors.

Note that RV-C is worse in that there is no way to change other parameters (I forget which).

jnm30327, 2023-03-09:
OK. Thanks!