kwindrem/GuiMods #141: Contributing to project?

stefprez, 2023-04-05:
Hello! Are you open to others contributing to this project? I'd like to start making some changes to some of the UI elements and see if I can add some features, but didn't see any more information about how to contribute to this project. Let me know if you're open to outside contributors, and how to best get involved.

kwindrem, 2023-04-05:
Your help could be useful. I would want to agree to any changes you add to the public GuiMods before incorporating them.

stefprez, 2023-04-06:
No problem. It'll take me some time to ramp up on the code base anyway and get familiar with everything. I might start off with cleaning up the README formatting. I think I read in another issue comment that your current workflow exists outside of Github. Are you open to letting Github manage versioning? I was noticing all the different version folders and the duplicative code it creates. Not something I'd try to tackle right away, but could help with clarity and getting additional community contributions.

kwindrem, 2023-04-06:
The versions are there to accommodate different Venus OS versions. This makes the mod run on multiple versions of Venus OS. I don't think GitHub could manage the type of versioning I'm doing. It allows the setup script to select the correct version of each modified file then install it in the operating directory. This is described in some detail here:

https://github.com/kwindrem/SetupHelper/blob/main/Package%20development%20guidelines.pdf

and more info in the SetupHelper updateFileSets script.

The majority of development is done on Mac OS using Xcode for editing mainly. I debug on a Raspberry PI running Venus OS. No IDE involved.

I'm using GitHub minimally but have moved the code base to a directory structure managed by GitHub.

stefprez, 2023-04-06:
Awesome, all helpful context. Thanks much! I definitely won't touch versioning/packaging anytime soon. I use Mac OS as well and primarily VS Code. I need to dig out my Raspberry Pi for this for sure. I have a Cerbo GX for my trailer, but I definitely don't want to be making that my dev environment and brick anything. I'll start digging in and will be in touch as I'm able to put some work into this. Cheers!

kwindrem, 2023-04-06:
I have a Cerbo in my travel trailer also. It gets used for development also because my PI doesn't have any devices attached. So I use Cerbo if I need to see actual numbers. The PI is great because I can reflash it if something I do bricks it.

SetupHelper does have tools to hopefully uninstall any mods I make even if you can't get to the GUI but still early development on Cerbo is risky.

mr-manuel, 2023-04-17:
I was also thinking of helping a bit to change the `ReadMe` files into `ReadMe.md` files with pictures. Is this ok for you? It would look much better!

kwindrem, 2023-04-17:
Yes, that's something I've been meaning to do. Didn't realize you could add pictures, but agree that's a good thing.

Some of the screen shots are out of date and the GuiMods Simple one is missing. So you would need to update the screen shots also. Then I think the screen shot files aren't necessary.

mr-manuel, 2023-04-17:
Check this [basic syntax](https://www.markdownguide.org/basic-syntax/). Should help you a lot. If you want the screenshot to only reside on GitHub, then you can add them directly. But for the `README.md` I would recommend a folder