kwindrem/GuiMods #142: No data displayed when using AC Meter and Charger State missing

pazu86, 2023-04-06:
Hi folks,

I'm just in the process of commissioning our Cerbo GX system and today I've hooked up an AC Meter (ET112), setup in 'Grid Meter' mode, I'm using Guimods which I'm loving - but I'm having an issue with it not displaying any info from the AC meter.

For context, this is a system on our boat, doesn't have an inverter, it is powered by a 12V system and 240v generator (Victron Phoenix charger, 2 x smart shunts etc... and the ET112 meter).

On the stock standard VRM Gui - I can see the AC Load/Input on the GUI fine.

[image]

On the Guimods Simple Screen, it also loads.
[image]

On the Guimods AC or DC coupled screen with the extra detail, I'm not seeing any of the AC load stats, so I'm not sure what I've configured wrong? 
[image]

Config screenshots below:
[image]
[image]

Further - I'm not seeing any state information on the AC Charger (Phoenix 50A/3 channel charger).
[image]
[image]

kwindrem, 2023-04-06:
Thanks for the reports.
I am beta testing some major changes to the way I gather information for the AC tiles. You can try this out by switching the branch/tag for GuiMods to "beta" in PackageManager. Let me know what changes if any. I'll look at the code difference between GuiMods Simple and DC/AC coupled pages also.

How are you getting information regarding the AC Charger? I've coded the AC Charger but you may be first to actually have one.

kwindrem, 2023-04-06:
I have made some changes to the way I enable AC tiles that should make them appear in your system. Please install the latest beta version and let me know what's shown.

I also fixed bugs in the way the DC System visibility is determined. This should make your DC system tile disappear from the simple flow page. You'll need to go to Settings / System setup and turn on Has DC system.

I made some changes in what devices are shown in the DC system detail page. I'm not sure if these will fix anything for you. Let me know if there are changes in what you see.

Still not sure why AC Charger isn't showing up correctly in the DC system detail page. I'll probably need the dbus info I identified in my above comment

Screen shots are helpful. Thanks.

pazu86, 2023-04-06:
Thanks for the speedy reply!

Ok, so I'm running Beta 7.30-8, assuming that is the right version?

Can confirm this does resolve the AC Inputs/Outputs, screenshot below.
[image]

Thank you! Super happy with that. 

One small note - the animation of the flow of power doesn't appear to work, you see it briefly coming out of AC input then nothing (not a big deal, just mentioning it).

On the DC System tile - that has disappeared from all screens now, AC/DC screens are the same - this is the DC screen for reference:
[image]

That said - I can get to the charger state info (at least one view) by tapping on the charger tile.

Can confirm that is still not working as per below screenshot:
[image]

For further reference, this is the status screen of the native menu when I go into the device from the main menu:
[image]

Being able to see all three batteries and states would be super helpful! Though I believe the charger itself can only be in the same state across all 3 batteries and any given time.

In terms of connection, the charger is just connected via a VE.Direct cable to the Cerbo - is that what you need or is there other info I can get to help?

Lastly - is it possible to show the charger stats in Amps (or both Watts and Amps)? I tend to use amps in the DC side and Watts in the AC side, not sure if that's a universal thing!

Thanks again for quick response - love the added functionality!

kwindrem, 2023-04-06:
Yes, that's the correct version. Glad the AC tiles are showing up now.

For the DC system tile to appear you need to go into Settings / System setup and turn on Has DC system.

I need more information on your charger. What is it exactly? How is ithe DC connected?

What are your batteries and how are they connected to Cerbo? smart shunts or ???

I really need to see the dbus info. For that you'll need console access (or ssh).

kwindrem, 2023-04-07:
Disregard my query about the AC Charger. I found the unit I think you are using: Phoenix Smart Charger -- the smallish box like the MPPT controllers. Is that correct?

At this point I don't need the dbus info.

The reason power wasn't appearing is the device doesn't report it (just voltage and current), so I'm adding code to calculate power.

State also had a typo. 

I'll be adding voltage and current to the AC Charger details page but not the DC System details as there isn't room.

I need to modify my DC System and AC Charger details page to display multiple AC charger outputs. Right now it only looks for a single output.

You are correct about state. It's the same state for all three batteries, which actually makes no sense but that's the way it's designed.

The system identifies one "system" battery, typically automatically selected but you can specify which one with System Setup / Battery monitor. This is the battery that appears on the flow pages. If there are other batteries in the system (need a shunt or a data connection to the battery), they will show up in the DC System details page under Other DC devices. 

This works on my system. If I select the inverter/charger as the battery monitor, then my smart shunt monitoring my battery appears in the Other DC devices list.

But your original screen shots show a system battery AND one called "Battery 1" in the Other DC devices list. Let me know what is selected as THE battery monitor and where Battery 1 might be coming from (maybe a second shunt?)

pazu86, 2023-04-07:
Thanks and that all makes sense, correct it the Phoenix Smart Charger - this one https://www.victronenergy.com/chargers/phoenix-smart-ip43-charger

Looks like an MPPT but the damn thing is huge. Photo next to Cerbo for context.

[image]

I believe the theory behind the single state is it uses the charger outputs to regulate it - so if one battery needs bulk charging, the others stop outputting any charge or drops right back as it charges the one that needs a top up, something like that.

So my current setup is as follows:

Main Battery 1 - Seperate Smart shunt, charged via Phoenix
Main Battery 2 - Seperate Smart shunt, charger via Phoenix
Generator Battery 3 - charged via Phoenix, no shunt as it's just for genset.

Voltage monitor is set on Automatic, which has selected Battery 2 as main monitor - as that is normally the one selected from the battery selector switch so has the most load, I can't actually select the charger as the monitor source interestingly.
[image]

Confirmed DC system working as well
[image]

Looking good!

kwindrem, 2023-04-07:
You should be able to name the two BMV-712s. Go into Devices for each of them and there should be a Name field. Might help you keep things straight.

Then you should be able to select the one you want rather than leaving it automatic. I think it'll use the first one it finds which could change so probably a good idea to not use automatic.

Working on other changes now. It's a big change so it may take a couple of days.

pazu86, 2023-04-07:
Yeah they actually are named, Battery 1 and 2, just not showing in the selection screen for some reason.

Does the changes you are planning include the Amps on the tiles? I didn't quite understand what you meant earlier - it's certainly not a big issue just a quality of life thing in terms of how I look at the DC side :) 

[image]

Really appreciate your help and work so far - makes the system way more useful!

kwindrem, 2023-04-07:
Amps will be shown on the AC Charger detail and the DC System detail. Amps will not be shown on the overview. That page is all watts for consistency and is the Victron ecosystem. A watt on AC side compares to a watt on the DC side where an amp on the AC side can't be compared to an amp on the DC side due to voltage differences. There are even some systems that have multiple battery banks running at different voltages.

Re batteries: Battery 2 has been chosen as the system battery so it won't show in the Other DC devices list on the DC system detail. But Battery 1 will show since it is not considered the system battery.

pazu86, 2023-04-07:
No dramas at all - makes sense, thanks very much :)

pazu86, 2023-04-07:
Thanks for the quick response again! So Guimods updated no dramas, and the state page has changed - though no data yet, is there anything else I need to configure?

[image]

Having them in the AC Charger detail would be great, though I also guess I can go into the main menu if needed.

kwindrem, 2023-04-07:
No nothing to configure unless you want gauges.
I just pushed a new beta (v7.30~10) that should show multiple charger outputs.

I'm not sure why no data is being displayed but power was a debug line I forgot to remove. 

Please try the new beta and if data is still not populating I would need the gui log file:

tail -50 /data/log/gui/current|tai64nlocal

If you don't have ssh or console access you can capture all log files using the Package Manager Backup function. That will place a file named logs.zip either on removable media or in /data

pazu86, 2023-04-07:
Awesome - it has updated, still though no data though, I do have SSH/Console access but not near boat at the moment, will need to go and pull it locally (unless you know of a way via VRM portal?) I need to look at an RPI or something as a vpn gateway in the network so I can hit it remotely.

Will try and get down there today.

[image]

kwindrem, 2023-04-08:
I found some typos and this should now be showing values in v7.30~11

pazu86, 2023-04-08:
Confirmed - see values now, but it looks like they are the same values for all 3 outputs - they are normally pretty close but slightly different. Two screens for comparison. Looking good though!

[image]

[image]

kwindrem, 2023-04-08:
Another fix (v7.30~12). Let me know.

pazu86, 2023-04-08:
bingo! Works perfectly thank you!

[image]

closed 2023-04-08 by kwindrem