kwindrem/GuiMods #143: GuiMods files set incomplete (3-Phase System)

kevstone122, 2023-04-06:
Hello and thanks for your help already!

I search already a lot for this error and there are a lot on Guthub, but not same like mine.
I have an 3-Phase system with 6 Multiplus and tried to install your GUI-Mods.

Downloaded the latest version from cerbo OS-large:
[image]

In the Package manger download the latest version from github and try to install it.
Always get fault and it doesn't work. Any Idea?
[image]

Thank you!

kwindrem, 2023-04-06:
This error occurs if the setup script can't find matching files for the current OS version. This can happen if one or more of the files is modified by something else.

If you have console access, I'd be interested in what is reported in /data/log/SetupHelper. If you don't have console access, you can gather all log files by using the backup and restore function within PackageManager. You can extract the SetupHelper log from the logs.zip or post the zip file here and I'll take a look.

kevstone122, 2023-04-07:
Hello Kwindrem!
Thank you for your fast answer. Very create what you provide here for support!
Attached the log direct copied with WinSCP.
[LOG SetupHelper.zip](https://github.com/kwindrem/GuiMods/files/11178513/LOG.SetupHelper.zip)

Thank you!

kwindrem, 2023-04-07:
There are 4 files the GuiMods setup script could not find replacements for. It's highly unusual for there to be more than one if something is modifying files.
I recommend the folloiwing:

Go to the PackageManager menu and then Active Packages then GuiMods and redownload and install the latest version which should be v7.21. 

If that does not fix your problem, then switch to the previous Venus OS version via the Stored firmware menu then reinstall the latest version using Online updates. This process forces the Venus OS to be reinstalled.

kevstone122, 2023-04-09:
Thanks for your feedback.
I download it in the menu again and again already. And it was 7.21 as you can see on the screenshot.
The Venus OS (Cerbo-GX) i update exactly before i start to install it.
So it will not change anything or do you modify anything for the download?

Thanks!

kwindrem, 2023-04-09:
No modifications, just the download then the install.

kevstone122, 2023-04-12:
I try again to download it in the menu, but doesn't work. 
I am afraid now, that anyway later on with any update i get problem....

Approved-OK, 2024-05-09:
Had the same Problem on osX 3.40-13

@40000000663c88601fbcfe0c GuiMods/setup: --- starting setup script v10.26 action INSTALL
@40000000663c886230c15c84 GuiMods/setup: ERROR v3.40~13 dbus_systemcalc.py no replacement file
@40000000663c88631eeea1ec GuiMods/setup: ERROR: incomplete file set for v3.40~13

althoug tried to deinstall and reinstall - did not help

kwindrem, 2024-05-09:
Try to download GuiMods again, then reinstall.

If that does not work try reinstalling the firmware: switch to the backup firmware then do an online update to the latest firmware. Don't just swap to the original version in the backup firmware menu.

Approved-OK, 2024-05-11:
> Thanks. I'll have a new version of GuiMods out in a few minutes that supports v3.40~13

it work's, v10.27 need only to be installed - all ok

closed 2024-05-11 by kwindrem