kwindrem/GuiMods #144: Ruuvi Barometric Pressure

dmac001, 2023-04-06:
Hi, Is it possible to add the Ruuvi tag barometric pressure to the various tiles?  Currently shows only temperature and humidity.  If space is an issue, allow select which items to display (e.g., temp + pressure).  Thank you

dmac001, 2023-05-08:
I modified "TileTemp.qml" , added the following lines to fetch the sensor pressure,
```
   property VBusItem pressureItem: VBusItem { id: pressureItem; bind: Utils.path(bindPrefix, "/Pressure") }
    property string pressureText: pressureItem.valid ? ("\n" + pressureItem.value.toFixed (0) + "mb") : ""
```
and and added pressureText to the tile display,

```
               else if (tempScale == 2)
                    return ((root.temperature * 9 / 5) + 32).toFixed (1) + "°F" + humidityText + pressureText
                else
                    return root.temperature.toFixed (1) + "°C" + humidityText + pressureText
```
Seems to work fine.  Please consider adding to GuiMods. 
Thanks
Don

dmac001, 2023-05-08:
Also, I modified PageTemperatureSensor.qml to display barometric pressure,

	`property variant summaryValue2: pressure.item.valid ? [pressure.item.text] : ""`

and

	title: getTitle()
	summary: temperature.item.valid ? summaryValue + ", " + summaryValue2 : status.valid ? status.text : "--"

For use on a sailboat, the barometric pressure is very important.  Please consider adding these 2 lines to  PageTemperatureSensor.qml.

Thanks,
Don

morphias2004, 2025-03-06:
+1

Can we please add pressure. I have just added a BME 280 to the i2C bus and it provides accurate temp, humidity and barometric pressure.