kwindrem/GuiMods #147: Temperature missing Alternator Detail screen

svalbard2, 2023-04-27:
Hi, I see that now the Temperature is missing on the Alternator Detail screen? I have the 7.41 build. RPM is there still. The temperature is visible in the detailed page of the Alternator (Regulator WS500).

kwindrem, 2023-05-01:
I think I fixed this. Please check v7.45 and report back.

svalbard2, 2023-05-06:
Hi, sorry. Still no temp on that page.

Regards Heinrich Eggenfellner

kwindrem, 2023-05-06:
I'll need screen shots of dbus-spy for your alternator to sort this out.

svalbard2, 2023-05-07:
I will see what I can do. Remember, it used to work before.. The regulator is WS500.

The temperature shows in the WS500 screen together with all other information. RPM, Field, etc.

svalbard2, 2023-05-07:
Have a look at the attached screenshots.

I am sorry I have no experience with dbus-spy. Can iot be run on a Windows computer?

Heinrich

kwindrem, 2023-05-07:
I looked at the code for the WakeSpeed menu page and I should be doing what it is.

I'm working on some changes anyway. No more info needed from you at this point.

I'll push out a beta later today for you to test.

FYI, dbus-spy runs on the GX device (aka Cerbo, etc) from the command line.

kwindrem, 2023-05-07:
switch the tag for GuiMods to "beta" and give my test code a try.

svalbard2, 2023-05-10:
Hi, now I have ver 7.48-1

Still not working?

kwindrem, 2023-05-10:
Try v7.48~2. Found a typo.

Sorry this is taking so long. I have nothing to check this code so I need to rely on reports from people like you that do have the device.

svalbard2, 2023-05-11:
Works!

Fantastic.

See attachments.

svalbard2, 2023-05-11:
[image]

svalbard2, 2023-05-11:
[image]

svalbard2, 2023-05-11:
[image]

svalbard2, 2023-05-11:
[image]

kwindrem, 2023-05-11:
Great. I'll be releasing the changes in a few minutes. Don't forget to set the branch/tag for GuiMods back to latest.

svalbard2, 2023-05-17:
Hi, I see i now have ver 7.49. The Temperature is still not working there? It worked in the previous Beta?

kwindrem, 2023-05-18:
I think I fixed it in v7.50. It looks like my merge from the beta branch didn't work

kwindrem, 2023-06-07:
v7.63 has another attempt to fix this

svalbard2, 2023-07-18:
Hi, now Guimods are incompatible with my CerboGX? Ver 7.72.

Something wrong?

Regards Heinrich