kwindrem/GuiMods #148: Guimods 7.41 incompatable with Cerbo GX v3.00~37

andyfoulger, 2023-04-28:
Updated to v3.00~37 and Guimods is showing incompatable.

thomasinaz41213, 2023-04-28:
ive got same problems. guimods will not install on 3.00~37

kwindrem, 2023-04-28:
Yup. Major change to generator logic in ~37. Need time to do some careful testing.

kwindrem, 2023-04-29:
should be working now. let me know if you experience problems, especially functionality with the new generator logic that includes warm-up and cool-down

kwindrem, 2023-04-29:
1.

closed 2023-04-29 by kwindrem

andyfoulger, 2023-05-01:
@kwindrem - Do you have any information on the new warm-up and cool-down feature? Wondering how it works.

kwindrem, 2023-05-01:
It's described in some detail in the beta release notes:

https://community.victronenergy.com/questions/202458/venus-os-v30037-available-for-beta-testing.html

Right now you must have the AC1 input type set to generator and AC2 input type NOT set to generator in order to see the warm-up and cool-down times. I think that's getting fixed.

Also right now, the input type needs to be set to generator in order for warm-up and cool-down to be included in the startup and shutdown process. My GuiMods or ExtTransferSwitch packages fix this.

When the generator starts, the generator AC input to the Multi/Quattro is set to an ignore state during the warm-up period. After that, the ignore is removed from the AC input allowing the generator to feed the Multi/Quattro.

A generator stop triggers a cool-down period in which the AC input is once again ignored. At the end of the cool-down, the generator is stopped and the ignore is removed.

thomasinaz41213, 2023-05-01:
Hey Kevin so to be clear, are you saying that the generator starts and then ac in is ignored for the warm up period?

kwindrem, 2023-05-01:
Yes, and ends that way too for cool-down.

Remember that the Multi/Quattro will take several seconds after generator power stabilizes to accept it and close the input relay, so I don't think there is any possibility that the generator would be loaded during the start-up process before the warm-up period has been satisfied.

thomasinaz41213, 2023-05-01:
I still seem to be having an issue. I have guimods 7.45, exttransferswitch 2.7 and venus 3.00~37.
The ac input tile is no longer changing between shore/generator and the warmup/cooldown is not functioning. (it shows warmup/cooldown on the generator tile, but the ac input is not disconnecting)
I tried uninstalling both guimods and exttransferswitch, reboot the cerbo, redownloaded and reinstalled both packages, then reboot again. I went through the exttransferswitch setup procedure again but no joy.
I don't know if there is still an issue somewhere, or maybe my system is being stubborn and some old data is not getting deleted/overwritten by the new?

kwindrem, 2023-05-01:
You need to set the grid source to shore or grid while the transfer switch is in the grid position. Also need to set the input current limit for grid and Don't start generator when AC present also. Then switch the digital input to generator and set the input current limit for generator. This is explained in the ExtTransferSwitch ReadMe.

You may need to update the firmware on your Multi. The beta release notes for v3.00~37 explain what's needed.

kwindrem, 2023-05-01:
Unisntall GuiMods and ExtTransfer switch and insure the actual transfer switch stays in the generator position. Set the AC 1 input to generator then run a maunual start and stop. If the Multi AC input still ignored during warm-up and cool-down then this is a bug in the Victron code somewhere. Post in answer in the beta test issue so Victron is aware and can look into it. Include your Multi model and the fact that is't two units in parallel.

thomasinaz41213, 2023-05-01:
It is working as expected without guimods or exttransferswitch installed. I then tried with each package individually and it does not function (warm up/cool down delay) properly with just guimods but it does function properly with just exttransferswitch. If I install both packages it does not work, regardless of the order in which I installed the packages.

kwindrem, 2023-05-03:
Great detective work. Looking into it now.

FYI, the two install scripts both attempt to install startstop.py, but have different functionality (GuiMods version does more). So I test for the GuiMods version in the ExtTransferSwtch install. This points to a problem in the GuiMods version.

kwindrem, 2023-05-03:
Should be fixed now with GuiMods v7.46.

thomasinaz41213, 2023-05-03:
Confirmed working as expected now. Great work Kevin. Appreciate all you do. You are a great asset to the community.

kwindrem, 2023-05-03:
Thanks. Accommodating an external transfer switch was of high interest to me even though I don't run a generator in my RV. (I do but it's manual start.) That and the warm-up/cool-down stuff have been requested by many. So I was glad to jump in.