kwindrem/GuiMods #151: Feature Request - Dark Mode

mr-manuel, 2023-05-12:
Hi,

since the [thread](https://community.victronenergy.com/questions/151889/feature-request-gx-touch-dark-mode.html) in the Victron forums seems a bit silent for many months I would propose to integrate a dark mode in your GuiMods.

I would open a PR for this, but first I wanted to make sure, if it‘s ok for you and ask you how you would integrate it.

My ideas were two:
1. Create a copy of all relevant QML files
2. Only search and replace the colors with a color map in the existing QML files

What are your thougts?

kwindrem, 2023-05-12:
Many have requested this so I'm sure it would be welcome.

You will need to create a preference in GuiMods menu for this and hook that up to all overviews. There is a section of code in the setup script that looks for one setting then updates/creates all settings. Your new setting MUST be the one that gets tested and also in the body of that test to actually create the setting.

I suggest you only do the "enhanced" overviews (the ones that GuiMods uses) and not try to do all stock overviews. There may be other files you'll need to add to the fileList and incorporate into the setup file.

Modify the existing replacement files, don't create additional ones. Hook the preference up to all files that affect screen color.

You will need to modify files in ALL file sets, not just the latest one. This provides compatibility for users not using the latest and greatest.

There are guidelines in SetupHelper for creating packages that should help you with your modifications to GuiMods.

mr-manuel, 2023-05-12:
The idea was to only add a switch in GuiMods. When the switch is triggered, then the `sed` command is run for only the relevant pages. I would only use the overview pages.

mr-manuel, 2023-05-12:
Ok, so basically I need to change all colors with variables in the QML files, so that they can be switched dynamically?

kwindrem, 2023-05-12:
Yes.

Use a setting like all other GuiMods options so the selection survives a restart and a firmware update.

mr-manuel, 2023-05-13:
How do I know, if I have to put the new QML files under the version file set or normal file set?

kwindrem, 2023-05-13:
the scrip updateFileSets creates file sets from stock versions stored in another location. I think this is explained in Package development guidelines in SetupHelper.

Files that do not change from version to version go in the FileSets directory and are not put in the fileList. Files that are version-dependent go in the version directories within the FileSets directory and are also listed in fileList. The setup script must also include lines to install and uninstall them.

mr-manuel, 2023-05-13:
I completed the changes but I have some troubles to install my beta test. I already created all files and so on, but it does not work as expected: https://github.com/mr-manuel/venus-os_GuiMods

Currently I'm stuck at:

```bash
2023-05-13 15:41:37.818026500 venus-os_GuiMods/setup: --- starting setup script v7.49
2023-05-13 15:41:37.969986500 venus-os_GuiMods/setup: creating file set for v3.00~42
2023-05-13 15:41:38.633172500 venus-os_GuiMods/setup: ERROR v3.00~42 PageMain.qml no replacement file
2023-05-13 15:41:38.689737500 venus-os_GuiMods/setup: ERROR v3.00~42 main.qml no replacement file
2023-05-13 15:41:38.734219500 venus-os_GuiMods/setup: ERROR v3.00~42 ObjectAcConnection.qml no replacement file
2023-05-13 15:41:38.785332500 venus-os_GuiMods/setup: ERROR v3.00~42 OverviewHubEnhanced.qml no replacement file
2023-05-13 15:41:38.856245500 venus-os_GuiMods/setup: ERROR v3.00~42 OverviewMobileEnhanced.qml no replacement file
2023-05-13 15:41:38.907386500 venus-os_GuiMods/setup: ERROR v3.00~42 TileTankEnhanced.qml no replacement file
2023-05-13 15:41:38.958462500 venus-os_GuiMods/setup: ERROR v3.00~42 MarqueeEnhanced.qml no replacement file
2023-05-13 15:41:39.001415500 venus-os_GuiMods/setup: ERROR v3.00~42 HubData.qml no replacement file
2023-05-13 15:41:39.058945500 venus-os_GuiMods/setup: ERROR v3.00~42 PageSettingsDisplay.qml no replacement file
2023-05-13 15:41:39.110366500 venus-os_GuiMods/setup: ERROR v3.00~42 Tile.qml no replacement file
2023-05-13 15:41:39.159627500 venus-os_GuiMods/setup: ERROR v3.00~42 PageSettingsRelay.qml no replacement file
2023-05-13 15:41:39.230254500 venus-os_GuiMods/setup: ERROR v3.00~42 PageSettingsGuiMods.qml no replacement file
2023-05-13 15:41:39.287551500 venus-os_GuiMods/setup: ERROR v3.00~42 TileRelay.qml no replacement file
2023-05-13 15:41:39.377028500 venus-os_GuiMods/setup: ERROR v3.00~42 OverviewTanksTempsDigInputs.qml no replacement file
2023-05-13 15:41:39.488241500 venus-os_GuiMods/setup: ERROR v3.00~42 dbus_systemcalc.py no replacement file
2023-05-13 15:41:39.546344500 venus-os_GuiMods/setup: ERROR v3.00~42 OverviewFlowComplex.qml no replacement file
2023-05-13 15:41:39.608217500 venus-os_GuiMods/setup: ERROR v3.00~42 OverviewGeneratorEnhanced.qml no replacement file
2023-05-13 15:41:39.666147500 venus-os_GuiMods/setup: ERROR v3.00~42 OverviewGeneratorRelayEnhanced.qml no replacement file
2023-05-13 15:41:39.735430500 venus-os_GuiMods/setup: ERROR v3.00~42 startstop.py no replacement file
2023-05-13 15:41:39.808818500 venus-os_GuiMods/setup: ERROR v3.00~42 dbus_generator.py no replacement file
2023-05-13 15:41:39.866559500 venus-os_GuiMods/setup: ERROR v3.00~42 PageGenerator.qml no replacement file
2023-05-13 15:41:39.933857500 venus-os_GuiMods/setup: ERROR v3.00~42 PageSettingsGenerator.qml no replacement file
2023-05-13 15:41:40.005824500 venus-os_GuiMods/setup: ERROR v3.00~42 TileDigIn.qml no replacement file
2023-05-13 15:41:40.041619500 venus-os_GuiMods/setup: ERROR v3.00~42 Battery.qml no replacement file
2023-05-13 15:41:40.077658500 venus-os_GuiMods/setup: ERROR v3.00~42 MbStyle.qml no replacement file
2023-05-13 15:41:40.113556500 venus-os_GuiMods/setup: ERROR v3.00~42 OverviewBox.qml no replacement file
2023-05-13 15:41:40.150493500 venus-os_GuiMods/setup: ERROR v3.00~42 OverviewConnection.qml no replacement file
2023-05-13 15:41:40.186474500 venus-os_GuiMods/setup: ERROR v3.00~42 OverviewConnectionEnd.qml no replacement file
2023-05-13 15:41:40.222645500 venus-os_GuiMods/setup: ERROR v3.00~42 OverviewGridParallel.qml no replacement file
2023-05-13 15:41:40.258624500 venus-os_GuiMods/setup: ERROR v3.00~42 OverviewHub.qml no replacement file
2023-05-13 15:41:40.310675500 venus-os_GuiMods/setup: ERROR v3.00~42 Tile.qml no replacement file
2023-05-13 15:41:40.348083500 venus-os_GuiMods/setup: ERROR v3.00~42 TileText.qml no replacement file
2023-05-13 15:41:40.384311500 venus-os_GuiMods/setup: ERROR v3.00~42 styles.css no replacement file
2023-05-13 15:41:40.414528500 venus-os_GuiMods/setup: ERROR: incomplete file set for v3.00~42 - can't continue
```

Yes it's explained, but I did not fully understand it. The big question for me is, how do I know, if the files changes from version to version? For some files I'm pretty sure they will not change, but some others I'm not sure.

Maybe you can take a short look? https://github.com/kwindrem/GuiMods/compare/main..mr-manuel:venus-os_GuiMods:main

kwindrem, 2023-05-13:
You can compare at each stock file in each version to see if there are differences but that's a lot of work.

Build the stock file sets then run SetupHelper/updateFileSets <package name> It will create the file sets and flag each replacement that does not exist. This runs on your HOST computer, not the GX device.

I use the raspberry pi images in the Victron feeds to grab /etc and /opt for each version.

mr-manuel, 2023-05-13:
Now I got it to work somehow. Can you send me somehow the `/etc`, `/opt` and `/var/www` folder of a fresh Venus OS before the breaking changes in `v3.00~42`? Unfortunately I do not have these files.

mr-manuel, 2023-05-13:
Here a few screenshots. I just have to fix that with the filesets and then it can be tested.
[image]
[image]
[image]
[image]
[image]
[image]

kwindrem, 2023-05-13:
Looks good. Just a couple of comments:

Most of the tiles stand out against the background but the flow lines appear a little dark. The battery "terminals' aldo but that's not so important. I wonder if the tiles should also be brightened a little also.

The remaining energy blue in the battery is very close to the background color so it looks like the text is falling off the bottom of the tile. But it's just the color. 

Is dark mode affecting all the menu pages also? If so, that’s great. I’m assuming this works on the local GUi as well as the remote console.

I noticed your AC in and out tiles are not showing frequency. Do you have meters on the in and out? If not, what inverter/charger are you using?

FYI, there is a hidden value for show inactive tiles. Use dbus-spy to set it to a value of 3 and the flow lines for inactive tiles will appear as well.

mr-manuel, 2023-05-14:
> Most of the tiles stand out against the background but the flow lines appear a little dark. The battery "terminals' aldo but that's not so important. I wonder if the tiles should also be brightened a little also.

I tried it in a dark environment (e.g. in a camper, van or boat at night). The goal was to produce as little light as possible, while seeing good the informations.

> The remaining energy blue in the battery is very close to the background color so it looks like the text is falling off the bottom of the tile. But it's just the color.

I will change that.

> Is dark mode affecting all the menu pages also? If so, that’s great. I’m assuming this works on the local GUi as well as the remote console.

Yes. It's affecting all:
[image]
[image]
[image]

> I noticed your AC in and out tiles are not showing frequency. Do you have meters on the in and out? If not, what inverter/charger are you using?

It's only a self written script, that is emulating a multiplus. The data for the grid meter is fetched from MQTT and there is no frequency data for [grid meters](https://github.com/victronenergy/venus/wiki/dbus#grid-and-genset-meter)? I would have the data available.

> FYI, there is a hidden value for show inactive tiles. Use dbus-spy to set it to a value of 3 and the flow lines for inactive tiles will appear as well.

Which value is it exactly?

I know there was a breaking change with `v3.00~42`. Can you send me the `/etc` and `/opt` files for the version before the braking changes? Since I have to add it for the file sets.

mr-manuel, 2023-05-14:
I wanted also to add a touch area on the overview pages to be able to fast switch between light/dark mode. Can you help me with that?

[image]

kwindrem, 2023-05-14:
The setting to show flow lines is:

com.victronenergy.settings /Settings/GuiMods/ShowInactiveFlowTiles

While it is not listed in the Victron document, there is a frequency parameter for grid/genset/output meters. One for each phase:

/Ac/L1/Frequency
/Ac/L2/Frequency
/Ac/L3/Frequency

/etc and /opt are almost 30 MB. You should be pulling these from the feeds. I use the .wic.gz files from http://updates.victronenergy.com/feeds/venus/candidate/images/raspberrypi4/ and http://updates.victronenergy.com/feeds/venus/release/images/raspberrypi4/

MacOS along with a ext4 extension mounts these as a disk image by changing the extension to .img. Not sure what the equivalent would be in the Windows world would be.

The changes from ~39 to ~41 were in the generator area. PageSettngsGenerator.qml is the only GUI file that changed.

I don't like the idea of hiding a light/dark switch as you suggest. It isn't obvious and when someone does it by accident they may not realize they can undo it. Mac OS has an automatic light/dark mode that uses time of day to make the change automatically. There is no empty real estate for a button on the overview pages. The switch could be moved to the Display & Language menu but that's only one menu  closer.

I suppose a help message like the "Tap tile center for detail at any time" might be acceptable but I wonder if there is a way to add a button to the menu bar that shows up when you tap the screen. Right now there is "Pages", "Menu" and up/down indicator for scrolling. That would make the light/dark button available in all overviews and menu pages.

mr-manuel, 2023-05-14:
Thanks!

Yes I know, that I can pull it from there, but only the last candidate and not the past versions. So I cannot create the version for the past Venus OS candidates... 

With Windows you can use 7-zip and just open the files without problems.

kwindrem, 2023-05-14:
Oh, right. Don't worry about older release candidates. By default updateFileSets only keeps the latest beta and only if there's no newer release. It marks older beta versions as UNUSED_FILE_SET but does not delete them. You can just delete the file set folder for those older betas. I've been keeping the more recent release candidates just because things are moving pretty quickly right now as Victron gets ready for the release of v3.00.

I'll be out of town and away from my development system the rest of this week so hopefully, v3.00 will be released before I merge your changes into main.

Thanks for working on this. It's a much asked for feature.

mr-manuel, 2023-05-14:
I don't know what I'm still doing wrong. Sometimes `/data/SetupHelper/updateFileSets venus-os_GuiMods` works, sometimes not.

My steps:

1. Creating a directory `/data/StockVenusOsFiles` that contains the data `/etc`, `/opt` and `/var/www` of Venus OS `v2.90`, `v2.91`, `v2.92`, `v2.93`, `v2.94` and `v3.00~42`.

3. Changed the path in `/data/SetupHelper/updateFileSets`
  ```
  # set these as appropriate to your system
  packageRoot="/data"
  stockFiles="$packageRoot/StockVenusOsFiles"
  ```

4. Added this files to the file list
  ```
  /opt/victronenergy/gui/qml/Battery.qml
  /opt/victronenergy/gui/qml/MbStyle.qml
  /opt/victronenergy/gui/qml/OverviewBox.qml
  /opt/victronenergy/gui/qml/OverviewConnection.qml
  /opt/victronenergy/gui/qml/OverviewConnectionEnd.qml
  /opt/victronenergy/gui/qml/OverviewGridParallel.qml
  /opt/victronenergy/gui/qml/OverviewHub.qml
  /opt/victronenergy/gui/qml/OverviewSolarCharger.qml
  /opt/victronenergy/gui/qml/OverviewSolarInverter.qml
  /opt/victronenergy/gui/qml/Tile.qml
  /opt/victronenergy/gui/qml/TileText.qml
  ```

5. Placed the files in the folder `/data/venus-os_GuiMods/FileSets/v3.00~42`.

Now when I run `/data/SetupHelper/updateFileSets venus-os_GuiMods` I get:
```
root@raspberrypi4:~# /data/SetupHelper/updateFileSets venus-os_GuiMods
Stock files path: /data/StockVenusOsFiles
venus-os_GuiMods: checking file sets
 - no original fileods: v3.00~42 PageMain.qml
 - no original so no replacement fileMain.qml
 - no original fileods: v3.00~42 main.qml
 - no original so no replacement file.qml
 - no original fileods: v3.00~42 ObjectAcConnection.qml
 - no original so no replacement filectAcConnection.qml
 - no original fileods: v3.00~42 OverviewHubEnhanced.qml
 - no original so no replacement fileviewHubEnhanced.qml
 - no original fileods: v3.00~42 OverviewMobileEnhanced.qml
 - no original so no replacement fileviewMobileEnhanced.qml
 - no original fileods: v3.00~42 TileTankEnhanced.qml
 - no original so no replacement fileTankEnhanced.qml
 - no original fileods: v3.00~42 MarqueeEnhanced.qml
 - no original so no replacement fileueeEnhanced.qml
 - no original fileods: v3.00~42 HubData.qml
 - no original so no replacement fileata.qml
 - no original fileods: v3.00~42 PageSettingsDisplay.qml
 - no original so no replacement fileSettingsDisplay.qml
 - no original fileods: v3.00~42 Tile.qml
 - no original so no replacement file.qml
 - no original fileods: v3.00~42 PageSettingsRelay.qml
 - no original so no replacement fileSettingsRelay.qml
 - no original fileods: v3.00~42 PageSettingsGuiMods.qml
 - no original so no replacement fileSettingsGuiMods.qml
 - no original fileods: v3.00~42 TileRelay.qml
 - no original so no replacement fileRelay.qml
 - no original fileods: v3.00~42 OverviewTanksTempsDigInputs.qml
 - no original so no replacement fileviewTanksTempsDigInputs.qml
 - no original fileods: v3.00~42 dbus_systemcalc.py
 - no original so no replacement file_systemcalc.py
 - no original fileods: v3.00~42 OverviewFlowComplex.qml
 - no original so no replacement fileviewFlowComplex.qml
 - no original fileods: v3.00~42 OverviewGeneratorEnhanced.qml
 - no original so no replacement fileviewGeneratorEnhanced.qml
 - no original fileods: v3.00~42 OverviewGeneratorRelayEnhanced.qml
 - no original so no replacement fileviewGeneratorRelayEnhanced.qml
 - no original fileods: v3.00~42 startstop.py
 - no original so no replacement filetstop.py
 - no original fileods: v3.00~42 dbus_generator.py
 - no original so no replacement file_generator.py
 - no original fileods: v3.00~42 PageGenerator.qml
 - no original so no replacement fileGenerator.qml
 - no original fileods: v3.00~42 PageSettingsGenerator.qml
 - no original so no replacement fileSettingsGenerator.qml
 - no original fileods: v3.00~42 TileDigIn.qml
 - no original so no replacement fileDigIn.qml
 - no original fileods: v3.00~42 Battery.qml
 - no original so no replacement fileery.qml
 - no original fileods: v3.00~42 MbStyle.qml
 - no original so no replacement fileyle.qml
 - no original fileods: v3.00~42 OverviewBox.qml
 - no original so no replacement fileviewBox.qml
 - no original fileods: v3.00~42 OverviewConnection.qml
 - no original so no replacement fileviewConnection.qml
 - no original fileods: v3.00~42 OverviewConnectionEnd.qml
 - no original so no replacement fileviewConnectionEnd.qml
 - no original fileods: v3.00~42 OverviewGridParallel.qml
 - no original so no replacement fileviewGridParallel.qml
 - no original fileods: v3.00~42 OverviewHub.qml
 - no original so no replacement fileviewHub.qml
 - no original fileods: v3.00~42 OverviewSolarCharger.qml
 - no original so no replacement fileviewSolarCharger.qml
 - no original fileods: v3.00~42 OverviewSolarInverter.qml
 - no original so no replacement fileviewSolarInverter.qml
 - no original fileods: v3.00~42 Tile.qml
 - no original so no replacement file.qml
 - no original fileods: v3.00~42 TileText.qml
```

I also tried to place the files only in `/data/venus-os_GuiMods/FileSets`.

This is now taking me more than a day for such a simple thing :-S red the PDF over and over again, but I don't get it.

kwindrem, 2023-05-14:
The messages are very odd. They don't seem to be formatted the way my logMessage function does.

For example, the call for the first message is logMessage "ERROR $package: $version1 $baseName - no original file"

Also, the second message call is logMessage "ERROR $package: $version1 $baseName - no original so no replacement file" but I'm seeing the file name at the end.

I run this script on Mac OS where I do my editing. I have never run this script on Venus OS so maybe there are differences. There could also be differences when running on Windows. Not sure where you are running it.

Some of the files creating errors should already be in the file set from my latest version. But even so, there should be stock files in /data/StockVenusOsFiles/v3.00~42.

mr-manuel, 2023-05-14:
I run it on a Raspberry Pi with Venus OS v3.00~42. I will try it on a VM with Debian Linux hopefully all runs fine there. Yes there are some files, that was making me questioning even more.

mr-manuel, 2023-05-15:
I now tried it also on Ubuntu and Fedore, but the same result. I used the SetupHelper `v4.40`. Maybe you could try it please? https://github.com/mr-manuel/venus-os_GuiMods/archive/refs/heads/main.zip

mr-manuel, 2023-05-15:
Damn. Found the issue. Somehow the line ending of `fileList` changed from `LF` to `CRLF` -.-"

kwindrem, 2023-05-16:
Glad you found the problem. I’ll see if I can make updateFileSets and CoommonResources more tolerant of fileList format.

I’m out of town for the rest of the week but will get back in touch after I get home

kwindrem, 2023-08-10:
mr-manual, an addition in v3.10 betas places an icon for wind and solar in the lower left corner of the battery icon. This works great in light mode, but the icon is nearly invisible in dark mode. Any suggestions as to what I should change to make fix this?

mr-manuel, 2023-08-11:
Have you a screenshot of it? Is there a way do display it without having a wind generator?

kwindrem, 2023-08-11:
Here's the screen shots in light and dark mode.
The functionality is enabled in the inverter/charger configuration (veconfig or VictronConnect). It doesn't actually require wind or solar components.

[image]

[image]

I'll admit that with the gauge, there's not much of the icon left.

I thought  about creating an alternate the icon file that is white rather than black.

kwindrem, 2023-08-11:
I was able to fix this myself. I dis make an alternate icon that was close to white that provides contrast with the darker tile:
[image]

closed 2023-08-11 by kwindrem