kwindrem/GuiMods #153: GeneratorConnector what happened?

bowpay, 2023-05-31:
Hello, this is Bowpay from victron forums.  I originally asked you to create the Generator Connector so that my external generator buttons triggered the GUI to show the generator was started and can be canceled.  What happened to it?  I have the GuiMods installed but it looks like the functionality is missing or I dont understand how to enable it.  It worked fantastic till i upgraded to 3 recently.

bowpay, 2023-05-31:
Link to external state!  Found the answer.  There are so many features I didn't initially see it.  Want to ask another request.  Is it possible to change the name of the AC2 input to Generator when external state is found and use Grid when no external state is connected?

kwindrem, 2023-06-01:
The functionality to switch between shore and generator is intended ONLY for AC 1 input. A Quattro has an internal transfer switch and one input should be set up for grid/shore and the other generator.

The original generator connector functionality should work regardless of which AC input is used for the generator. The time accumulation is based on the state of the generator present digital input. I did add functionality that looks at AC input being generator if there's no digital input.

Things will get messed up if you put a grid/generator transfer switch on AC 2 input and set it to generator in system setup.

bowpay, 2023-06-01:
AC1 is our main 50amp.  AC2 is designated for generator and shore power.  It uses a transfer switch.  It was designed that way so that we can use the control dial inside to adjust the amperage.  It works great for RV parks at 50amp and when we moochdock with friends 20amp or less.  Just dial it down and we dont have to worry about popping breakers or, in the case of AC1, we dont have to change anything.  I was thinking that if the digital input for generator, when signal is supplied, could change the AC2 from shore power to generator under Settings > System Setup > AC Input 2.

bowpay, 2023-06-01:
After reading your reply again I think we are on the same page.  I would set AC2 to shore power initially.

bowpay, 2023-06-01:
Btw we have two separate cables one for 50 and the other is a 30amp line.  We spend 75% of our time on 50amp side, 15% on generator and 10% mooch docking.  I didnt want to adjust ac1 everytime we got on the road.  Also the generator is rated at 8kw so I would have to adjust it everytime we ran the generator.

kwindrem, 2023-06-01:
It would be so much simpler from a system standpoint to use one input for shore power and the other for generator. That's the way Victron designed the system. Yes it means changing the input current limit to match your shore power but that's one setting per setup and you still need to do that if you aren't using the 50 amp shore source.

Be that as it may, I have two separate components that have both been added to GuiMods.

One is the generator interface that accepts a digital input to tell Venus OS that the generator is running. That interlocks with manual start. Auto start conditions flag an error if the generator is not running. I recently added the ability to use AC input presence as a substitute to the digital input. That works OK as long as you always have the AC input configured for generator. But if you are using a transfer switch ahead of that input, the system can't determine if the generator is running if the transfer switch reports grid power and assumes the generator is not running when it might be. Any time accumulation would not include such periods. So it is essential to use the "generator is running"  digital input when using an external transfer switch or otherwise changing an AC input from grid to generator.

The section component is support for an external transfer switch. This was designed for Multis with only one AC input so you can switch between grid/shore and generator power with another digital input.

I would have to modify things to work with a Quattro with the transfer switch on AC 2 in. I think it would work as-is if the transfer switch were on the AC 1 in.

bowpay, 2023-06-01:
I wanted to point out that we run a 240v 8kw quattro with two auto formers.  One steps up power after the transfer switch and the second drops voltage on the output.  It's been awhile but I believe it was designed that way due to the 240v and auto formers to eliminate the need for two inverter/chargers and to reduce the overall size of the equipment.  If my memory serves me correctly the 30amp side needs to be stepped up to 240v with the auto former and the 50amp can go directly to the quattro.  If its possible to consolidate the 50/30 sides then I would but I am thinking its not possible.

kwindrem, 2023-06-01:
If your generator is 120 volt I can see why you did it that way. I'll look into modifying ExtTransferSwitch to work on AC 2 in.

bowpay, 2023-06-01:
Yea, its a Onan Diesel 8kw 120v.

kwindrem, 2023-06-03:
I have updated ExtTransferSwitch to support the transfer switch connected to Quattro AC 2 In. I didn't find any changes needed to GuiMods to support this change. 

I do recommend you do not use the AC input presence to determine if the generator is running since you have the transfer switch ahead of that AC input. This makes generator running inaccurate. Use a dedicated digital input to tell the GX device if the generator is running or not.

closed 2023-06-03 by kwindrem

bowpay, 2023-06-04:
Awesome you rock!  Btw is it bundled with GuiMods or do i need to manually install it?

kwindrem, 2023-06-04:
GeneratorConnector functionality was merged into GuiMods with additional enhancements that were not in GeneratorConnector. ExtTransferSwitch is still separate and needs to be installed/updated independent of GuiMods.

bowpay, 2023-06-04:
I dont see it when i go to package manager.

kwindrem, 2023-06-04:
You will not see GeneratorConnector but should see GuiMods and ExtTransferSwitch. If not, check the Inactive packages list. You can add the package from there.

If ExtTransferSwitch is not in either list, perform Initialize PackageManager ...

bowpay, 2023-06-04:
I Initialized but still not seeing it.  Also still see Generator Connector under inactive packages.

bowpay, 2023-06-04:
Looks like after clicking initialize the package manager responds with package manager not running.

kwindrem, 2023-06-04:
Try rebooting.

Also, look in the package manager log:

/data/log/PackageManager/current

GeneratorConnector will appear but you should not install it if you are running GuiMods.

bowpay, 2023-06-04:
root@einstein:/data/log/PackageManager# cat current
@40000000647cb411218c66dc WARNING:>>>> PackageManager v4.41 starting
@40000000647cb4f725c05844 WARNING:received PackageManager INITIALIZE request from GUI
@40000000647cb4f822a90df4 WARNING:stopping threads
@40000000647cb4f823dd9a3c WARNING:shutting down com.victronenergy.packageManager dbus service
@40000000647cb4f828264c24 CRITICAL:>>>> PackageManager exiting
@40000000647cb4fa021bc1bc WARNING:>>>> PackageManager v4.41 starting
@40000000647cb4fa0e24990c WARNING:adding GuiMods
@40000000647cb4fa17be839c WARNING:adding SetupHelper
root@einstein:/data/log/PackageManager#

kwindrem, 2023-06-04:
OOPS. My error. I never added ExtTransferSwitch to the default package list. I'll fix that, but in the mean time you can add it manually.

Go to Inactive packages and tap on new. Fill in all the info for it:

Package name: ExtTransferSwitch
GitHub user: kwindrem
GitHub branch/tag: latest

then add the package. It should then show up in Active packages list. From there you can download and install it.

bowpay, 2023-06-04:
Got it setup!  I'll test it this week and let u know.  Any eta on including transfer switch in package manager.  I reinitialized and lost it so i had to readd it.