kwindrem/GuiMods #156: Displaying Motordrive Detail view

iain-mothership, 2023-06-06:
I am trying to test out the motor drive support by simulating the Sevcon motordrive using PCAN prior to getting the Curtis controller to emulate the Sevcon messages.   So far I can get the top level to display the power parameter but have had no luck trying to get the detail screen to display any of the other data.  I have looked through the code/readme and don't see any configuration other than the max power parameter.  Here is the dbus-spy data and the screenshots.  Happy to dig in to debug but wanted to see if I was missing anything obvious first.

```
com.victronenergy.motordrive.sevcon0
Controller/Temperature                                   35
Dc/0/Current                                         32.125
Dc/0/Power                                             1542
Motor/RPM                                               114
Motor/Temperature                                        57
```
[image]

[image]

kwindrem, 2023-06-06:
I've looked over my code and can not explain the problem. Is there anything in the GUI's log that indicates a problem?

tail -50 /data/log/gui/current | tai64nlocal

iain-mothership, 2023-06-06:
Just tried to show the detailed motor drive data and no new errors.  I had been changing some parameters and restarting earlier which may be the cause of some of the earlier errors (local time according to the cerbo is 15:47 when I last tried to view the detailed view).

btw - which file has the top level tile definition for the Motor Drive?

```
2023-06-06 13:40:53.331145500 "using /etc/venus for runtime features" 
2023-06-06 13:40:53.331154500 "running on einstein" 
2023-06-06 13:40:53.430871500 Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
2023-06-06 13:40:53.939063500 Creating settings 
2023-06-06 13:40:54.923281500 [VeQItemDbusPublisher] Registered service "debug.victronenergy.gui" 
2023-06-06 13:40:54.923299500 language is "en" 
2023-06-06 13:40:55.097545500 accessLevelChanged:  2 
2023-06-06 13:40:55.097553500 setAccessLevel:  2 
2023-06-06 13:40:55.111938500 LocalThemeDaemonClient: Looking for assets in "/opt/victronenergy/themes/ccgx" 
2023-06-06 13:40:55.111949500 Screen: 1024x600 150x90mm 173x169dpi
2023-06-06 13:40:55.409023500 [Qwacs::qwacsRegisterd]  "com.victronenergy.qwacs" 
2023-06-06 13:40:55.549952500 [Qwacs::qwacsRegisterd] service not registerd  "com.victronenergy.qwacs" 
2023-06-06 13:40:56.283672500 TimeZone::valueChanged:  "/UTC" 
2023-06-06 13:40:56.329175500 ScreenSaver::dbusDisplayOffTimeChanged:  600 
2023-06-06 13:40:56.329183500 ScreenSaver::enable 
2023-06-06 13:40:56.329185500 ScreenSaver::screenOn 
2023-06-06 13:40:56.329187500 Buzzer::setBuzzer: false 
2023-06-06 13:40:56.329189500 ignoring request for "dbus/com.victronenergy.system/Buzzer/State" to set value QVariant(bool, false) (not online) 
2023-06-06 13:41:00.309129500 Get Items failed "com.victronenergy.motordrive.sevcon0" 
2023-06-06 13:41:00.409822500 [Service] DBus service "/service/websockify-c" changed: "VncLocal" - "1" 
2023-06-06 13:41:00.410146500 [Service] installing "/service/websockify-c" 
2023-06-06 13:41:00.527337500 [Service] DBus service "/service/websockify-c" changed: "VncLocal" - "1" 
2023-06-06 13:41:00.563620500 [Qwacs::qwacsRegisterd]  "com.victronenergy.qwacs" 
2023-06-06 13:41:00.753670500 Get Items failed "com.victronenergy.qwacs" 
2023-06-06 14:05:52.868060500 ScreenSaver::timer 
2023-06-06 14:05:52.868257500 ScreenSaver::screenOff 
2023-06-06 14:09:44.469614500 VePlatformVenus::wakeupEvent 
2023-06-06 14:09:44.469811500 ScreenSaver::enable 
2023-06-06 14:09:44.469816500 ScreenSaver::screenOn 
2023-06-06 14:09:44.470813500 VeQuickView::mousePressEvent: wakeupEvent 
2023-06-06 14:22:54.089344500 ScreenSaver::timer 
2023-06-06 14:22:54.089352500 ScreenSaver::screenOff 
2023-06-06 14:35:43.150105500 VePlatformVenus::wakeupEvent 
2023-06-06 14:35:43.150307500 ScreenSaver::enable 
2023-06-06 14:35:43.150312500 ScreenSaver::screenOn 
2023-06-06 14:35:43.151534500 VeQuickView::mousePressEvent: wakeupEvent 
2023-06-06 14:45:56.337791500 ScreenSaver::timer 
2023-06-06 14:45:56.337988500 ScreenSaver::screenOff 
2023-06-06 14:49:37.311377500 VePlatformVenus::wakeupEvent 
2023-06-06 14:49:37.311609500 ScreenSaver::enable 
2023-06-06 14:49:37.311615500 ScreenSaver::screenOn 
2023-06-06 14:49:37.312642500 VeQuickView::mousePressEvent: wakeupEvent 
2023-06-06 15:15:26.817220500 ScreenSaver::timer 
2023-06-06 15:15:26.817412500 ScreenSaver::screenOff 
2023-06-06 15:17:35.182230500 VePlatformVenus::wakeupEvent 
2023-06-06 15:17:35.182240500 ScreenSaver::enable 
2023-06-06 15:17:35.182242500 ScreenSaver::screenOn 
2023-06-06 15:17:35.183320500 VeQuickView::mousePressEvent: wakeupEvent 
2023-06-06 15:25:04.300568500 service  "/net/connman/service/wifi_48e7da859e1d_hidden_none"  was removed but was not found in the services list 
2023-06-06 15:32:32.114164500 service  "/net/connman/service/wifi_48e7da859e1d_hidden_none"  was removed but was not found in the services list
```

iain-mothership, 2023-06-06:
> The motor drive detail is DetailMotorDrive.qml which uses DcSystemRow.qml for the rows in the table.
> DcSystemRow.qml is common to several detail pages and seems to work in other areas.
> DetailMotorDrive.qml uses case DBusService.DBUS_SERVICE_MOTOR_DRIVE to find motor drives to place in the table. This is the same filter used to set up the entry in the Device List. You might check to see if your device is showing up in the Device List.

Just checked and it doesn’t appear on the top level device list or the ve.can devices list which explains the problem This is using the built in motor-drive driver (which the standard Venus build pulls from https://bitbucket.org/oceanvolt/dbus_motordrive/src/master/).  This hasn’t been updated for years and links to an old velib so I guess nitrite has happened. Will see if I can get victron to take a lookThanks for your prompt attentionSent from my iPhone

kwindrem, 2023-06-06:
It's odd. The GUI uses the DBusService. ... but systemcalc (which creates the sum shown on the overview) uses the first part of the service name.

closed 2023-10-16 by kwindrem