kwindrem/GuiMods #157: Temperatures from Mopeka tank sensors

helmet1978, 2023-06-06:
Would it be possible to have the temperature readings from Mopeka wireless tank sensors available for display, either in the tank detail or as "regular" temperature sensors? They are available via MQTT under tank/##/Temperature

kwindrem, 2023-06-06:
If a tank sensor shows up on dbus, I'll show it. If the Mopeka tank sensors are not showing up on dbus, then work needs to be done at a lower level to support them. This is something Victron would need to do.

closed 2023-06-06 by kwindrem

kwindrem, 2023-06-06:
According to this:

https://community.victronenergy.com/questions/148366/mopeka-pro-sensors-in-venus-29018.html

The LP and H2O sensors are already supported. They must be the Pro version though. Have you tried them?

Eagle383, 2023-06-06:
I use the mopeka sensors. They work well you can see the temps in the sub menu and on the config page. Just not on the overlay. 
[image]

helmet1978, 2023-06-06:
Yes, these are all the Pro version and they look to be working fine, was just wondering about the temperatures. Thank you!