kwindrem/GuiMods #159: Missing PV data for Multi RS Solar?

mfu-mcosys, 2023-06-14:
Maybe it is a Problem because i use a Quattro (just inverting) and a MultiRS (just charge) on the same Cerbo... 
VenusOS is 3.00

[image]
[image]

kwindrem, 2023-06-14:
It has something to do with the Multi RS MPPT. Is this a multi-tracker device?

Do you have command line access to your Cerbo? If so, I'd like to see a screen shot of the dbus info for the RS.
If not, at least a screen shot of the device's menu.

mfu-mcosys, 2023-06-14:
I have access through ssh on the cerbo, but which logfile is for the MultiRS?
This is the list of dbus logfiles in /var/log/

dbus-adc
dbus-ble-sensors
dbus-digitalinputs
dbus-fronius
dbus-generator-starter
dbus-modbus-client
dbus-modbustcp
dbus-mqtt
dbus-qwacs
dbus-shelly
dbus-systemcalc-py
dbus-vebus-to-pvinverter
gps-dbus.ttyUSB0
gps-dbus.ttyUSB1
gps-dbus.ttyUSB2
mk2_dbus.ttyS4
smartbms-dbus.ttyUSB0
smartbms-dbus.ttyUSB1
smartbms-dbus.ttyUSB2
vecan-dbus.can0

kwindrem, 2023-06-14:
Not a log file but an app that shows the dbus parameters:

dbus-spy

Then select the RS and it will show you all the parameters. There may be more than fits on a screen.

I'm guessing there's something wrong with my logic that looks for multiple MPPT trackers

mfu-mcosys, 2023-06-14:
Hope this helps a bit, maybe some info from the mqtt helps too.

[image]
[image]

kwindrem, 2023-06-14:
It does help. Thanks. Your RS is a single tracker so my original though was not correct.

I'm now thinking that because this a combined inverter/charger and MPPT combined, I'm not recognizing it as a PV charger. Working on changes now. Stay tuned.

kwindrem, 2023-06-14:
I just published v7.64 as a fix for this. Please give it a try and report back.

mfu-mcosys, 2023-06-15:
Looks nice
[image]
[image]

Thanks a lot!

I know the AC-stuff isnt related to the "PV Charger"-Tile, but for my usage i want the combined PV/AC power for charging in the "PV Charger"-Tile. Lets see where i need to change some vars. :)

closed 2023-06-15 by kwindrem

mfu-mcosys, 2024-06-16:
seems like after the update to 10.42 today the informations on the flow are missed again :/