kwindrem/GuiMods #160: GuiMods and communication with v3.00

Ivo12s, 2023-06-22:
Using simple configuration - external LiFePo4 battery with Victron BMV700 and standard shunt. 2x BlueSolar Charger MPPT 150/60 and Smart Phoenix Inverter 48V 5000VA 230V. GuiMods installed on Raspberry Pi4B and version 2.94. In the configuration I setting up "have DC system", where it is standard 99-118 W as continuous calculated DC load.

When I installed version 3.0, the DC Load calculated total consumption included AC load. AC Load highlighted as standard. So for simple definition presented:
Before - with v.2.94:
AC Load - 1095 W
DC Power - 118W

After - with 3.0:
AC Load - 1095 W
Solar - 1148 W
DC Power - 2275 W

The same in VRM Portal and Remote Console as well.
When downgrade into 2.94, everything goes into normal.
[image]

kwindrem, 2023-06-22:
Does this problem still exist if you completely uninstall GuiMods?

kwindrem, 2023-06-22:
I looked over the code and found some errors I introduced in calculating DC System power. These should be fixed in v7.65

closed 2023-06-22 by kwindrem

Ivo12s, 2023-06-23:
👍😎