kwindrem/GuiMods #164: Background is light on web display in Dark Mode

svScoots, 2023-07-08:
I am using an android phone for the display for an octoGX.  I'm loving the dark mode, (I'm on a sailboat) but the border around the active area and the background around the virtual hotkeys are both still white. Can you make the surround area dark, and dim the virtual hotkeys a bit as well? I've attached a screenshot of how it currently looks on my system.  
Venus 3.00 and GuiMods 7.68

[image]

kwindrem, 2023-07-08:
This excellent work was done by @mr-manuel. I recall him saying that the areas you speak of are controlled by the device's light/dark mode, not the one selected in the GX device's menu.

It works on Safari on Mac OS, but not on my iPhone, and apparently not on your Android phone either. 

Hopefully this is something easy that he can fix.

mr-manuel, 2023-07-08:
Hi, I‘m using an iPad and Android Tablet and it works on both, when I enable the device dark mode. I‘m using on both devices  Firefox and Brave.

kwindrem, 2023-07-08:
Yup, seems to work on my iPad as well. So... there's something specific about the phones. This gets into the twisty maze of web site design which I'm not at all familiar with. Not sure anything can be done although I know others have posted ideas for removing the hotkeys section altogether.

mr-manuel, 2023-07-08:
I found out, that on smaller resolutions there is another theme applied. Which phone do you have and which resolution, with which screen zoom and font size (android settings)?

kwindrem, 2023-07-08:
I have an iPhone 13 Pro and using the default text size (whatever that is). Looks like this:

[image]

svScoots, 2023-07-09:
I went back and tried a bunch of browsers, it works correctly in Windows 10 using Chrome in Dark mode, both the background and the hotkey background are dark and the keys are dim.  

I put Chrome in Android in dark mode and the background went dark but the hotkey background  stayed light like the above screenshot.  I put the phone in forced dark mode and there was no change.  The Brave browser on Android was the same.

mr-manuel, 2023-07-09:
Desktop mode changes, because there are two different button sets. I will take a look at it the coming week.

mr-manuel, 2023-07-10:
Since the hotkey background was different on devices with a width smaller than 795px it was displayed white. This will be fixed with the next version.

[image]

[image]

mr-manuel, 2023-07-10:
The changes are in the main branch. The cache of the Safari browser on the mobile devices is a bit tricky. Did you try in a Private tab?

kwindrem, 2023-07-10:
Clearing the cache worked. Thanks. I'll publish this to latest now.

closed 2023-07-10 by kwindrem

svScoots, 2023-07-18:
Great stuff! thanks everyone