kwindrem/GuiMods #166: Flow Chart Screen - missing status, missing LEDs

aph70, 2023-07-10:
Hi all,
don't know if this issue is GuiMods related, but if not you may have the knowdlege of where do dig for the rootcause.

Past weekend I did switch from my Raspi based solution to a version with a new Cerbo GX.

For the sake of simplicity, all devices are still connected to the Cerbo via USB (VE.Direct and MK3-USB), i.e. no direct VE.Bus connection.

The problem:
The status of the Multi as well as the LEDs are no longer displayed in the flow chart screen. 
It is always "Off" or "---" even if the inverter is "On", set to "Invert only" or "Charge Only" even though it exaclty doing that.
Also the LEDs are also no longer displayed. 
It worked flowlessly  with the Raspi under the same firmwares.

[image]
[image]
[image]
[image]

Already tried the following:

- Cerbo firmware downgarde to 3.0
- UI switched to English
- Turned off all power  and rebooted
- Gui mods reinstalled
- 

My system:

- Cerbo GX, FW: v.3.10~12 with Gui Mods v7.68 , SetupHelper 4.42
- GX Tank 140
- Ruuvi tags
- MultiPlus Compact 12/1600/70-16, FW: 505
- SmartShunt 500A, FW: v4.14
- Smart Solar Charger MPPT 75/15, FW: v1.61

Any ideas and thanks in advance!

Arne

kwindrem, 2023-07-10:
It appears your Multi is not being recognized. by your Cerbo. To confirm this, look in the Device List. If so, this has nothing to do with GuiMods.

If the Multi IS showing up in the device list, try uninstalling GuiMods. If the Multi appears in the overview page then the problem is not with GuiMods.

Let me know.

aph70, 2023-07-10:
All devices including the Multi do show up in the device list:
[image]

I also can set the Mulit to Load, Invert  or whatever and it will show the correct status.
[image]

It also states that it is connected via VE.Bus.
[image]

GuiMods removed via Setup-Helper I can do power limitation and it shows the selected AC-Modus:
[image]
[image]

But still the Flow-Screen doesnt show correct status, nor LEDs nor the bullets for power flow (if seen the power flow bullets missing though before when on remote console).

[image]

kwindrem, 2023-07-10:
From the last screen shot, it's apparent that the GX device is not receiving AC and inverter status (off, on, inverting, etc) from the Multi. Also, this is not an issue created by GuiMods.

I'm assuming you have tried rebooting Cerbo and power cycling the Multi.

While using the MK3-USB adapter should be working, it is possible the problem is with using this connection rather than a direct VE.Bus connection between the Multi and Cerbo.

kwindrem, 2023-07-10:
I can confirm there is a problem connecting the Multi via the MK3-USB on Cerbo.

I will report this as a bug to Victron.

closed 2023-07-10 by kwindrem