kwindrem/GuiMods #167: Font Change

BossAbj, 2023-07-10:
Hello, would it be possible to change fonts so those of us with 3 phase systems can see most of the data in a neater format?

kwindrem, 2023-07-10:
What do you have in mind? I feel the small fonts I use for more than a single phase is about as small as it could be and still be readable.

The only approach would be to remove some of the information (e.g., not show voltage and current like the stock tiles do).

kwindrem, 2023-07-10:
I found a solution that I think is much better:
[image]

comments?

BossAbj, 2023-07-10:
> I found a solution that I think is much better:
> 
> [image]
> 
> 
> 
> comments?
> 
> 
> 
> 

Awesome …. This looks very clean, how can I test it, so I give you some feedback ? 

[image]In my present one the wording for limit isn’t there , I love how you wrote it out and moved it down and left frequency centred above , perfect

kwindrem, 2023-07-10:
These changes are released now. Just do an update and the changes will appear.

I had no space for "Limit" when it was combined on the same row as the frequency.
Since there is no longer a space between L1 and L2 the frequency and input current limit can stay on two lines like it is for a single phase system (just a font size change).

BossAbj, 2023-07-10:
> These changes are released now. Just do an update and the changes will appear.
>
>
>
> I had no space for "Limit" when it was combined on the same row as the frequency.
>
> Since there is no longer a space [...]

Thank you sooooo much Kevin
You are truly the best !!!
Presentation is excellent now.
I’m always available for beta testing if you need me 

Here it is 

Can I plead for one more modification, the temp values show an increment of .5 for my Quattro but not for BMV, is it possible for them to just be rounded up , so we have 17 or 18 instead of 17.5 … ? 

A pic of my setup below as well [image]

[image]

kwindrem, 2023-07-10:
I am using common formatting for all numeric values and settled on 3 significant digits based on several other users wishing to see as much resolution as possible. (Some want to see wattage to 5 significant digits even though there's not that much precision in the system. They then complain when things don't add up to the watt!!!)

If I was to truncate the pecision to the nearest degree, I'd get complaints. Short of adding yet another user preference there's not much I can do to satisfy everyone.

BossAbj, 2023-07-10:
> I am using common formatting for all numeric values and settled on 3 significant digits based on several other users wishing to see as much resolution as possible. (Some want to see wattage to 5 [...]

I understand perfectly , you are far too kind 🙏🏾

BossAbj, 2023-07-10:
> I am using common formatting for all numeric values and settled on 3 significant digits based on several other users wishing to see as much resolution as possible. (Some want to see wattage to 5 [...]

Is there any article you can point me to , dummy startup to solar panels, mppt and all for Victrons, about to move to solar and need enough education, so I can tackle it myself . Want to go offgrid but at least start with 20/30 panels

BossAbj, 2023-07-10:
> I am using common formatting for all numeric values and settled on 3 significant digits based on several other users wishing to see as much resolution as possible. (Some want to see wattage to 5 [...]

I thought of this , is it possible to put the default battery values in the circled portion , so even if I remove batteries from the bottom view , the battery temp value used by the battery icon will always show either bmv or Quattro depending on what the user set in system setup ?

[image]

kwindrem, 2023-07-10:
I have a little personal experience with solar on a small scale (3 200 watt panels on my RV). I really don't know of any good texts on the subject. I have come across some info on the web that describes differences between PWM and MPPT for example. That should be easy to find. 

In terms of implementation, you can either connect panels through an MPPT controller to the battery bank or use a grid-tie PV inverter on either the input or output side of the inverter-charger. There are rules you need to follow if you put it on the output. Basically, no more solar than the size of your inverter. This is because the way the inverter controls PV output is to increase frequency and that takes time. Before that can happen, the inverter and batteries need to absorb the PV energy when a load suddenly removes itself from the output. You can find details about the "1 to 1 rule" in Victron documentation. PV inverters on the input side of the inverter only work if the grid is present. You can actually combine all three PV locations (AC in, AC out, battery).

One thing that needs careful research is sizing your array. You can't simply add up the specified power of the panels you use as your system output. Angle to the sun plays an important part in how much power you actually get.

Victron provides a calculator to help size MPPT controllers to an array.

kwindrem, 2023-07-10:
The space you identify is used for remaining battery time which is only shown when the battery is discharging.

BossAbj, 2023-07-10:
> I have a little personal experience with solar on a small scale (3 200 watt panels on my RV). I really don't know of any good texts on the subject. I have come across some info on the web that [...]

Thank you for the suggestion, won’t be doing a grid tie though since we don’t have that grid set up yet in my country , mppt to battery bank for me , already bought the 450/100, and trying to get my welder to come frame a support that takes into consideration the sun position thru out the day like you pointed out . 

Hopefully the panels can carry some portion of the load in the morning , will do more research 

Thank you

I forgot the battery to go time 💔😃

BossAbj, 2023-07-11:
> The space you identify is used for remaining battery time which is only shown when the battery is discharging.

Nice touch with remote login landing page , dark mode support is welcome 

Still on the subject of GUI unification , I see the suggestion on removing the space is implemented on the battery read out ,54.37V vs 54.37 V, I also believe the FC system should use same font unification by removing the space 63.2 W can be 63.2W, in the same breath all value in grid and AC loads can get closer and the space gained used in better separation, for example ; 
L1: 1470W  245V  9.0A
Looks cleaner and well balanced 
This can go for all variables in grid ; AC out and any other panes. 

Why I thought of this , was as I looked at the battery value 54.37V  -2.0A, I realised the font can reduce to the size of font above , so we have 54.37V -2.0A 15°C on one line .

The idea is to have the default basic layout as pictured without a cramped Home Screen and yet be able to see all basic important data at a glance 

How do you see it ?

Because even the Victron stock theme doesn’t have spacing between the values and SI units as pictured as well [image]

[image]

kwindrem, 2023-07-11:
I added the space between the number and the unit to improve readability.

Battery voltage precision was a specific request by several users.

BossAbj, 2023-07-11:
> I added the space between the number and the unit to improve readability.
> 
> 
> 
> Battery voltage precision was a specific request by several users.

Perfectly understandable for readability, if per chance and menu option can be added to remove the space if it’s not too much to ask , it would help those of us with multi systems , so data is still neatly presented .

Thank you 🙏🏾

BossAbj, 2023-07-24:
Good morning Kevin, sorry to bother you, just wanted to ask if there’s any hope with the menu item to remove spacing for those with multi 3 phase systems ? Thank you

kwindrem, 2023-07-24:
Not going to happen.