kwindrem/GuiMods #169: Relay functionality request

lfhoward, 2023-07-12:
Dear Kevin,

Having six working relays is a great upgrade from the basic Victron Venus OS. I would love it if relays 3-6 were able to be programmed similarly to relay 2!

Right now on my Raspberry Pi, relay 1 is the generator dialog. I coopted it to schedule when I want the battery heater to be on in winter (6 am-2 pm). If there were a more generic "scheduler" function that is not for the generator per se, but could be custom named, that would be even better. It would need a manual on and off to override the schedule to be perfect.

Relay 2 uses the temperature function along with data from RpiTemperature to turn the Raspberry Pi's cooling fan on and off.

I would love to be able to use relay 3 in the same way, but using data from my RuuviTag temperature sensor to turn the camper's ceiling fan on and off. This would keep the trailer's interior cool in the hot sun. However, relays 3-6 are not programmable at the moment. Can you make them so?

I think with bluetooth sensors like RuuviTags and Mopeka sensors becoming integrated with Venus, it makes sense to be able to use data from them to make more decisions, like being able to trigger multiple relays. 

Thanks a ton for all your work and for this fantastic software package!
Lauren

kwindrem, 2023-07-12:
I have considered expanding the functionality of the other relays and looked into it some. It's not easy but doable. Every use of a relay needs to scan all available relays for its function then use it if found. That's a lot of code to modify.

thomasinaz41213, 2023-07-12:
> Dear Kevin,
> 
> Having six working relays is a great upgrade from the basic Victron Venus OS. I would love it if relays 3-6 were able to be programmed similarly to relay 2!
> 
> Right now on my Raspberry Pi, relay 1 is the generator dialog. I coopted it to schedule when I want the battery heater to be on in winter (6 am-2 pm). If there were a more generic "scheduler" function that is not for the generator per se, but could be custom named, that would be even better. It would need a manual on and off to override the schedule to be perfect.
> 
> Relay 2 uses the temperature function along with data from RpiTemperature to turn the Raspberry Pi's cooling fan on and off.
> 
> I would love to be able to use relay 3 in the same way, but using data from my RuuviTag temperature sensor to turn the camper's ceiling fan on and off. This would keep the trailer's interior cool in the hot sun. 
> 
> Similarly, Relay 4 could turn on a cabinet fan to keep the battery compartment cool in summer, using temperature data from a RuuviTag in there.
> 
> However, relays 3-6 are not programmable at the moment. Can you make them so?
> 
> I think with bluetooth sensors like RuuviTags and Mopeka sensors becoming integrated with Venus, it makes sense to be able to use data from them to make more decisions, like being able to trigger multiple relays. 
> 
> Thanks a ton for all your work and for this fantastic software package!
> Lauren

You can do this with node red. Until such time as this becomes possible thru the gui, node red is probably the best option.

lfhoward, 2023-07-12:
> I have considered expanding the functionality of the other relays and looked into it some. It's not easy but doable. Every use of a relay needs to scan all available relays for its function then use it if found. That's a lot of code to modify.

@kwindrem, Thanks for considering programming this when you have the chance! I'm sure I am not the only one thinking that having additional functional relays would rock.

@thomasinaz41213, Thanks for that info. I have held off going for the large install of Victron's Venus OS because my trailer is pretty small and I want to keep the system streamlined and simple to conserve battery power. Have you noticed a difference in power consumption going to the large OS with Node Red, compared to the smaller installation without it?

thomasinaz41213, 2023-07-12:
> @kwindrem, Thanks for considering programming this when you have the chance! I'm sure I am not the only one thinking that having additional functional relays would rock.
>
> @thomasinaz41213, Thanks [...]

I can't say if it's made any difference in power consumption but I imagine it would be negligible if any. I make pretty extensive use of node red and have a lot of flows doing a lot of things, as well as a lot of dashboard gauges and controls. I haven't noticed any detrimental effect.