kwindrem/GuiMods #170: Current limit on gen on ac2

bowpay, 2023-07-16:
I got the generator and 30amp working on AC2.  Thanks for making the change!  I am now running into an issue where I can't change the current limit if the AC2 isnt enough wattage and the batteries go below 50% SOC and the generator kicks on.  I've got the amperage set to 10 (220v) and i need to change the amperage to 20 once the generator has kicked in.  Is this possible?  I do have a control panel but don't mind abandoning it if it means i can automatically setting based on AC input/generator.

kwindrem, 2023-07-16:
I'm not sure why you can not change the AC current limit for the ACTIVE source. That's pretty much automatic in the Victron code and I have not changed it.

You do need "overruled by remote" enabled in the Multi/Quattro setup (veconfig). There are separate settings for AC 1 and AC 2 inputs for a Quattro.

Remind me about your setup. 

ExtTransferSwitch stores the ACTIVE AC current limit for the current source before changing to generator. Then if you make current limit changes once generator is active then that setting is then updated in the ExtTransferSwitch storage for use the next time generator becomes active.

bowpay, 2023-07-16:
Sorry, I should have explained better.  I am the one with the 240v quattro AC1 on a 50amp line, AC2 running an ext transfer switch between a 8kw 120v gen and a 30amp line that we sometimes dial down for a 20amp house line.  The problem we have is remote monitoring.  Were going through extreme heat right now and I am running AES assist.  Once the batteries drop below 50% SOC the gen kicks on but the current limit panel is set to 10amps so the gen won't pull any more than the house line is pulling since the current limit is set to 10amps.

kwindrem, 2023-07-16:
This should already be working. While on 30 amp shore power, set the AC input current limit to the desired value (10 amps). Switch to generator and once the system has stabilized on generator set the AC input current limit to whatever you need for generator (15 amps). When you switch back to 30 amp shore power, the input current limit should reset to 10 amps. The next time the generator starts, the input current limit will again change to 15 amps. When you connect to 50 amp shore power (AC 1), the input current limit should switch to whatever value was active when that input was active (50 amps).

bowpay, 2023-07-16:
It looks like i cant go below 11amps when the control panel is set to override?  11amps translates to 22amps 2 more than the allowed 20amp breaker I am on.  Is there something I am missing?  Also I misspoke I am not using AES only using power assist.

bowpay, 2023-07-16:
I have a bit of a mess here.  I just kicked on the generator and, with the ExtTransferSwitch installed, the UI is not registering AC2 as Generator on the main screen.  I had the AC2 current limit on the physical control panel set to 11amp.  I then turned the generator on and set it to 20amp.  After turning off the generator AC2 came back showing shore still however the 20amp stayed on the control panel.  

[image]

[image]

As you can see the generator is showing as connected in the settings but the main menu just shows Shore power.

bowpay, 2023-07-16:
BTW I have AC2 set to Shore Power.

kwindrem, 2023-07-16:
I added an item to the Setup page for the external transfer switch digital input. You must set this to AC 2 in

kwindrem, 2023-07-16:
There's a document that describes the minimum ac current limit per product here:

https://professional.victronenergy.com/downloads/firmware/file-download/?name=VE.Bus%20%28Multis%2C%20Inverters%20and%20Quattros%29/Minimum%20input%20current%20limits%202022-12-07.xlsx

These are the values for firmware 423 and newer. Prior to that the numbers were significantly greater. If your Quattro's firmware is older you may be bumping up agains that. I'm also not sure how the DMC interacts with changes made in the GX device.

bowpay, 2023-07-16:
I'm running 502.  Weird the doc says 10amp yet my control panel won't go lower than 11.  The current limit has been such a PITA when it comes to being on household lines.  I also have the "On Generator" set under External AC Input Transfer Switch set.  Am I missing another setting?

[image]

kwindrem, 2023-07-16:
Yes. Device list / External AC Input transfer switch / Settings / External transfer switch connection

This should appear for Quattros and is hidden for Multis.

bowpay, 2023-07-16:
I see Setup under Device List / Exernal AC Input transfer switch /Setup

[image]

bowpay, 2023-07-17:
So it looks like having the control panel connected messes things up.  I have since disconnected the panel and its now letting me set it at 10amps.  I still can't get the main screen to display Generator when connected.  It also won't go back to the current I had set on the AC2 input.  Right now its set to 20amp on generator.  I then shutdown the generator and the ext transfer switch kicks in the shore power but the 20amp current limit is still set and will pop the breaker.  I must be missing a setting here.  I updated all the packages just in case to the latest.  Just to make sure we are talking about the same thing...  I need to change the current limit from 20amp to 10amp when the ext transfer switch goes to shore power.  I then want the current limit to set back to 20amp once the generator takes over the AC2 line again.  Everything is happening on AC2.   AC1 is dedicated for a 50amp @ 25amps (not overruled by the remote).  

As of now when I leave in a few minutes if the generator shuts off due to reaching the 95% SOC tonight then the current limit will remain at 20amps.  At that point the 20amp line will pop the breaker and I will have to come back to the RV tonight to reset the breaker.  This is a major PITA I've been doing this every day for the past 3 weeks.  We are set to leave in 2 weeks but getting this resolved would be a HUGE help in my gas and time bill.  Can I please buy you a beer to help me resolve this?

bowpay, 2023-07-17:
I am also assuming that the main screen would reflect "Generator" because the ExtTransferSwitch swaps the connection.  Is that a correct assumption? 

[image]

kwindrem, 2023-07-17:
Yes, your assumption is correct. The AC input tile should be switching between Shore and Generator. If this is not happening, then the input current limit won't change either.

Further, your setting for External transfer switch connection appears to be correct.

There are a couple of log entries of interest, both in /data/log/ExternalTransferSwitch/current

The first is discovered Quattro at or discovered Multi at. If the Quattro is not being detected then the AC 2 input won't be controlled regardless of the setting.

The second is Transfer switch is on AC 2 in or Transfer switch is on AC 1 in.

For your setup it should report Quattro and AC 2 in

Let me know what you find. We'll get this sorted ASAP.

kwindrem, 2023-07-17:
Also let me know if there are any ERROR lines

bowpay, 2023-07-17:
Yes I have it all setup on AC2.  I don't have a ExternalTransferSwitch folder.  My folder is labeled ExtTransferSwitch (probably doesn't matter).  Here is the log file contents...

@4000000064acc4281a998454 INFO:root:waiting for settings
@4000000064acc4291bef67c4 INFO:root:waiting for settings
@4000000064acc42a1c31470c INFO:root:waiting for settings
@4000000064acc43037acb174 INFO:root:Ext Transfer Switch digital input type valu                                          >
@4000000064acc43923023a14 INFO:root:discovered switch digital input service at                                           >
@4000000064acc439248b8994 Traceback (most recent call last):
@4000000064acc439248bb0a4   File "/data/ExtTransferSwitch/ExtTransferSwitch.py"                                          >
@4000000064acc439248bbc5c     self.getVeBusObjects ()
@4000000064acc439248bc42c   File "/data/ExtTransferSwitch/ExtTransferSwitch.py"                                          >
@4000000064acc439248bd3cc     if vebusService == "---":
@4000000064acc439248bdb9c UnboundLocalError: local variable 'vebusService' refe                                          >
@4000000064af2f0c05c881c4 *** CCGX booted (0) ***
@4000000064af2f1527a694d4 INFO:root:>>>>>>>>>>>>>>>> ExtTransferSwitch starting                                          >
@4000000064af2f1529a5a0f4 INFO:root:waiting for settings
@4000000064af2f162a1c058c INFO:root:waiting for settings
@4000000064af2f172a5fe874 INFO:root:waiting for settings
@4000000064af2f182aa770f4 INFO:root:waiting for settings
@4000000064af2f1e3a00a00c INFO:root:Ext Transfer Switch digital input type valu                                          >
@4000000064af2f25142be6fc INFO:root:discovered switch digital input service at                                           >
@4000000064af2f2517b1d19c Traceback (most recent call last):
@4000000064af2f2517b1fc94   File "/data/ExtTransferSwitch/ExtTransferSwitch.py"                                          >
@4000000064af2f2517b20c34     self.getVeBusObjects ()
@4000000064af2f2517b21404   File "/data/ExtTransferSwitch/ExtTransferSwitch.py"                                          >
@4000000064af2f2517b21fbc     if vebusService == "---":
@4000000064af2f2517b2278c UnboundLocalError: local variable 'vebusService' refe                                          >
@4000000064b451280eafae84 *** CCGX booted (0) ***
@4000000064b45131392d5634 INFO:root:>>>>>>>>>>>>>>>> ExtTransferSwitch starting                                          >
@4000000064b45132005c0c74 INFO:root:waiting for settings
@4000000064b451330167f27c INFO:root:waiting for settings
@4000000064b4513401a983a4 INFO:root:waiting for settings
@4000000064b4513501efeb14 INFO:root:waiting for settings
@4000000064b4513a0f329e0c INFO:root:Ext Transfer Switch digital input type valu          
>
@4000000064b4513f34216004 INFO:root:discovered switch digital input service at                                           >
@4000000064b4513f37a9bf8c Traceback (most recent call last):
@4000000064b4513f37a9e2b4   File "/data/ExtTransferSwitch/ExtTransferSwitch.py"                                          >
@4000000064b4513f37a9f254     self.getVeBusObjects ()
@4000000064b4513f37a9fa24   File "/data/ExtTransferSwitch/ExtTransferSwitch.py"                                          >
@4000000064b4513f37aa09c4     if vebusService == "---":
@4000000064b4513f37aa1194 UnboundLocalError: local variable 'vebusService' refe                                          >
@4000000064b46ee009ef9a1c ExtTransferSwitch/setup: --- starting setup script v2                                          >
@4000000064b46ee02d8f8554 ExtTransferSwitch/setup: ++ Installing ExtTransferSwi                                          >
@4000000064b46ee03457f0c4 ExtTransferSwitch/setup: restarting ExtTransferSwitch                                          >
@4000000064b46ee0364fd8c4 ExtTransferSwitch/setup: restarting dbus-generator-st                                          >
@4000000064b46ee1000e9c3c ExtTransferSwitch/setup: restarting dbus-digitalinput                                          >
@4000000064b46ee10e1625d4 ExtTransferSwitch/setup: completed

kwindrem, 2023-07-17:
Dang, I thought I'd fixed that particular error. I'll look into it.

kwindrem, 2023-07-17:
Progress. Looks like you were on an older version but up to date now. Appears the service is running. But I don't see the Transfer switch is on AC ... message, so maybe there's more to do.

Is the system working now?

I've made some changes that should make things more robust so the service won't crash, and with additional error logging. v2.19 is available on 'latest'. Give that one a try.

bowpay, 2023-07-17:
I updated and its working now!  The one thing though that I saw was that the Generator screen is still displaying "Running by manual start" after I turn it off the generator using the manual button in the RV.  Is there a setting to force it to reevaluate the signal?  

[image]

kwindrem, 2023-07-17:
Glad things are working now.

The stock code does not look for generator running state externally. But there is a notification if the system thinks the generator should be running . However, GuiMods does if you turn on

Device list / Settings / Generator start/stop / Settings / Link to external running state

The recommended way to provide running state is via a digital input set to Generator. If that input isn't set up, then AC presence is used if one of the inputs is of type Generator. When an external transfer switch is used, the generator may be running even if the AC input is switched to shore power. Therefore, the digital input is highly recommended. Alternatively, a AC input meter configured for Generator can provide stable running state. 

When my modified generator code sees a change in internal generator running state and the generator isn't running due to an automatic run condition, it brings manual start into sync with that status. If an automatic run condition is active, no change in run state occurs. However, the enhanced generator overview will flag the fact that the generator should be running but is not due to external influences. The stock generator overview will not display external running status.

bowpay, 2023-07-17:
I have link to external running state enabled.  I don't have the detect generator at AC Input, if that matters.  I have the digital input set to External Transfer Switch and running my generator on to the digital input 1.  How would I go about setting two digital inputs and generator/ExtTransferSwitch?  

[image]
[image]

kwindrem, 2023-07-17:
Detect generator at AC input is only used for the alarm.

I think the problem is that as soon as the external transfer switch goes back to grid, the generator running state can no longer be detected via the AC input.

Set up another digital input to Generator and use the same signal you are using to indicate transfer switch on generator.

bowpay, 2023-07-18:
That worked!  I bridged the digital input 1 and 2 and set the digital input 2 to generator.  It now recognizes the Generator when its shutdown using a manual button.  Thanks again man this is better than ever.  Where do I send that beer too?

kwindrem, 2023-07-18:
Glad it all worked out. Are you using the enhanced generator overview? It also provides access to auto start enable, generator service counter and adds "Running or Stopped" in the generator icon for a little additional indication that the generator is running.

closed 2023-07-18 by kwindrem

bowpay, 2023-07-18:
Yes but I've had an issue with fat fingering the auto enable so I just left it with the default one so we don't accidently disable the auto start.  lol