kwindrem/GuiMods #173: AC Loads in Mobile view shows only L1 information in a 3-phase system

dfch, 2023-07-27:
There is a discrepancy in GuidMods for a three phase setup regarding "AC Loads" information shown between the "Mobile" view and "Simple Overview" and "Tiles" views.

**Current behaviour**

In "Mobile" overview, only the first phase *L1* of all three phases *L1, L2, L3* is shown:

[image]

[image]

**Comparison to other views**

In "Simple Overview" and "Tiles" views I can see information about all three phases *L1, L2, L3":

[image]

[image]

**Expected behaviour**

I would expect the complete information of all three phases shown in the "Mobile" view (as shown in the "Simple Overview").

[image]

**Version running**

I am running GuiMods v7.75 on Venus OS v3.01 (on a Victron Cerbo GX).

kwindrem, 2023-07-27:
A limitation, more than a discrepancy. Obviously there is no space for three phase info in the Mobile Overview AC tiles. The same is true for the stock Mobile Overview. In fact, the stock version only shows total power.

I could probably reduce the font sizes and make room for a second phase but to show three the fonts would be very tiny.

dfch, 2023-07-27:
I agree. But, instead of showing only 2 of the 3 phases, I would then rather show only Ampère for each phase plus the frequency/Hertz (but leave out voltage and individual Watts per phase). But just showing L1 is a little bit irritating.
But pls do not get me wrong. I can totally live with the current implementation.