kwindrem/GuiMods #175: One DC System device displayed but multiple present

proptied, 2023-07-29:
Firmware: `v3.10~21`
SetupHelper: `v4.42`
GuiMods: `v7.75`
ExtTransferSwitch: `v2.24`

I have multiple DC System devices installed, but only one device is listed on the "DC System Detail" screen.

Is this page supposed to list each of them in a row, like the "other DC devices" table?

**Page:**
[image]

**Devices:**
[image]

[image]

**DC System 1 ("24V Source")**
[image]

**DC System 2 ("24V Load")**
[image]

kwindrem, 2023-07-30:
Yes, I would expect the two .dcsystem. devices would appear in the upper part of the table.

It is possible the second row in the system table is hidden. See if you can scroll the table to it by swiping. If so, I will attempt to adjust the table sizing logic.

proptied, 2023-07-30:
Ah! You nailed it... I swiped in the upper table and it scrolled to the second one. Thanks!

kwindrem, 2023-07-30:
I had a bug in the logic that sizes the tables. Should be fixed in v7.77

closed 2023-07-30 by kwindrem