kwindrem/GuiMods #176: Vorzeichen der Anzeige ändern

Sprocki38, 2023-07-30:
Hi,
Habe einen Smart - Shunt korrekt angeschlossen.
Aber wenn die Akkus geladen werden habe ich ein Minuszeichen vor den Werten im Cerbo GX.
Umgekehrt wird bei Entladen kein Zeichen angezeigt.
Gibt es die Möglichkeit einer Werte-invertierung im Setup zu integrieren?
Gruß
Gerd

kwindrem, 2023-07-30:
I have a SmartShunt in my system and the power and current signs are correct (- for discharging).

Sprocki38, 2023-07-30:
Dann bleibt das ein Rätsel.

kocki2, 2023-11-11:
Moin Gerd,
meint du die Übersichtsseite?
Falls ja, bei mir wird es richtig angezeigt, Fotos zeigen Ladezustand ...
[image]
[image]
Eine Möglichkeit kann sein, dass du am SmartShunt die Anschlüsse Battery und load vertausch hast.
LG
Gerd

Sprocki38, 2023-11-11:
Hallo Gerd, Der Shunt ist richtig angeschlossen. Hier ein Screenshot von den Werten

Und passend dazu meine Übersicht Ich habe im Setting das DC-System aktiviert. Allerdings habe ich bei Victron kein DC- System als Hardware gefunden. Im dunkelgrünen Feld müsste ja eigentlich der Wert auch negativ angezeigt werden. BG Gerd

Sprocki38, 2023-11-23:
Da sind wohl die Bilder nicht angekommen. 
Neuer Versuch 
[image]
[image]