kwindrem/GuiMods #177: Relay 1-Generator start/stop

Miniozzy3, 2023-07-31:
GUIMODS 7.77 running latest 3.20~21
generator will start when relay closes, it will never open until I go into relay settings and switch to manual.  Then relay opens.
removed GUIMODS and generator works as it should (Start/stop on command).  
This just started today, I'm sure it was working.  Today was the first time I let it get to SOC low level before autostarting.

I'm not sure what else to check.  Generator screen showed "stopped" once SOC high level was reached.

kwindrem, 2023-07-31:
GuiMods did make some changes in this area as part of v7.77 but I can not reproduce this issue here, so I need a bit more information.

Just to confirm, you are saying that the STATUS tile in the generator overview says "Stopped" but the relay remains closed?

Do you have a cool-down period set? If so how long?
Same for warm-up?

Are you using an external transfer switch?

Are you using my ExtTransferSwitch package?

If not, are you switching the AC input type to "Generator"?

Are you starting and stopping the generator manually or via the auto start mechanism?

Try a manual start/stop if you have not done so already.

If you have access to the system logs, I'd be interested in the last few lines that show the shutdown process:

tail -50 /data/log/*gen*/current | tai64nlocal

If you don't have ssh access, you can copy all the logs to a USB flash drive using PackageManager's backup settings mechanism. You can then post the results here.

Miniozzy3, 2023-07-31:
Well I reinstalled 7.77. Started generator (manually). Waited until it recognized incoming power. Generator was listed as power source.  Now generator will shutdown manually again.  
I do not have an external transfer switch.  So I have never installed the transfer switch pkg.

And I can confirm that the status screen (tile) was showing stopped.  Relay would not open when it was supposed to shutdown.
I would have to go to relay settings and switch relay 1 to manual and it would then open the relay.

I'm going to re-run the exact circumstances I did yesterday.
ran A/C until SOC activated the auto-start.
charged batteries back to 100%.  Generator should shutdown but didn't
(status tile in generator screen showed stopped)

I will try and get a log for you once I have performed test.

Miniozzy3, 2023-07-31:
So it had to be just an issue with updating to 7.77.  once I re-installed and rebooted a couple times.  My generator issues are gone.  I let the batteries get down to SOC generator start twice.  Both times the generator shutdown once 100% charge was achieved.

If no one has thanked you recently, for all the hard work you have put in over the years.  Let me say it  "Thank You"

Jason Cleave

closed 2023-07-31 by Miniozzy3