kwindrem/GuiMods #178: 20230717162050 v3.01 GUIMods 7.77 incompatible

albundy0815, 2023-08-06:
Hi, today I've updated to 3.01 (to be honest, I don't know how to find out the 3.01~xx)
I can only see the date of the release and GUIMods shows me, that it's incompatible. Is this already known?
[image]

albundy0815, 2023-08-06:
I had a look on:
https://github.com/kwindrem/GuiMods/issues/119
and this works for me as well. Uninstalled GUIMods, reboot and installed again.

Nevertheless... could please someone give me a hint how to see the Build-Number? I mean the Venus OS v3.xx~**%%**

kwindrem, 2023-08-06:
A number following ~ indicates a pre-release (otherwise know as release candidate) version. v3.01 is released so there will be no ~ and no number following it.

closed 2023-08-06 by kwindrem