kwindrem/GuiMods #179: Installed but no setup helper

paulmc28, 2023-08-13:
I just did a blind install, package manager is installed but it didn't install setup helper by default. It is listed in the inactive packages, but if I try to activate it nothing happens.

kwindrem, 2023-08-13:
That's very strange. If you are seeing PackageManager in the menus, then SetupHelper is definitely installed, or at least the PackageManager service is still present from a previous install.

If you have not done so already, try rebooting.

If that does not work, scroll to the bottom of the Package manager menu and select Initialize PackageManager and do the Initialize.

paulmc28, 2023-08-13:
Tried it. Just hangs at ...initializing and restarting.

kwindrem, 2023-08-13:
Did you reboot after the initialize?

Next would be to try the blind install again. Make sure you remove the USB stick when the GX device BEGINS its reboot.

Last thing to try would be a blind uninstall, then a blind install of SetupHelper.

paulmc28, 2023-08-13:
I have tried all of this multiple times.

The first time I installed setuphelper it was in the active list. Then I mistakenly removed it and it has not worked since.

Thanks for all the help.

kwindrem, 2023-08-14:
I found a bug in Package manager. should be fixed now in SetupHelper v4.43. You will probably need to do a blind install again.

closed 2023-08-14 by kwindrem

paulmc28, 2023-08-14:
Ran the new install, same results.

kwindrem, 2023-08-14:
Did you download the new blind install from GitHub?

paulmc28, 2023-08-15:
I did.

kwindrem, 2023-08-15:
I will need to see the logs in order to help further. Unfortunately without PackageManager running you'll need ssh access to the GX device to gather the log files.