kwindrem/GuiMods #186: White/blank GUI screen morning after Venus 3.01 update

lfhoward, 2023-09-01:
Hi Kevin,

I upgraded to Venus 3.01 yesterday and all seemed to be well at the time. This morning the HDMI screen was blank. I was able to get into the Venus console on my phone over wifi, and I could click the return button and access the menus. Once inside the menu, the HDMI screen displayed the menu properly and behaved normally. Clicking on Pages returned me to a blank screen with no touch functionality. 

I am running Venus 3.01 on a Raspberry Pi 3b. 
Rebooting Venus did not fix it. 
Reverting to Venus 3.00 did not fix it. 
I confirmed the version of GuiMods I am running is 8.0 and I have it set to update automatically. 

The /data/log/gui/current file is full of these errors:
@4000000064f20b4c1f774a4c unknown mouse event > type=4, code=5, value=0

I hope this is enough information to find a fix. If not, let me know and I can find you the data you need. I can SSH into Venus. 

Thanks a ton,
Lauren

lfhoward, 2023-09-01:
I want to add a couple of details:

I am able to swipe left and right on the Pages screen and the Generator & Relay pages DO work! It is the main display page that is blank. 

I also installed Louisvdw’s Dbus-SerialBattery software yesterday. This worked fine in the GUI yesterday, and still works (in the menu). I don’t know if this installation would be relevant.

kwindrem, 2023-09-01:
I’ll have to dig into this mor but am in the road today. You can set the branch/version for GitHub in Package Manager. Change latest to v7.81 to put GuoMods back a version--Kevin

lfhoward, 2023-09-01:
I will try that! Thank you.

lfhoward, 2023-09-01:
Reverting to v7.81 worked. I have a battery screen again. Thank you.

kwindrem, 2023-09-01:
Great. Sorry this happened. I’ll fix this ASAP but will need to be tomorrow. If you can snag the gui log file it will save me a step. No worries if that’s a hassle--Kevin

lfhoward, 2023-09-01:
The log file currently reads:

@4000000064f228ef0165047c QVNCServer new incoming connection from "127.0.0.1"
@4000000064f228ef17c3fa0c QVNCServer Client protocol version RFB 003.003
@4000000064f228ef17c4194c
@4000000064f228ef17ca4f24 QVNCServer No password configured, accepting connection
@4000000064f228f82301c0fc QVNCServer disconnected from "127.0.0.1"

A previously saved one reads:

@4000000064f227041b1ee0f4 unknown mouse event type=4, code=5, value=0
@4000000064f2270526d7c76c unknown mouse event type=4, code=5, value=0
@4000000064f2270534de5aec unknown mouse event type=4, code=5, value=0
@4000000064f2270534e0471c unknown mouse event type=4, code=5, value=0
@4000000064f2270534e88094 unknown mouse event type=4, code=5, value=0
@4000000064f2270534edef34 unknown mouse event type=4, code=5, value=0
@4000000064f22705350beee4 unknown mouse event type=4, code=5, value=0
@4000000064f22705350df66c unknown mouse event type=4, code=5, value=0
@4000000064f227053510ccfc unknown mouse event type=4, code=5, value=0
@4000000064f2270535119c04 unknown mouse event type=4, code=5, value=0
@4000000064f227053512ebf4 unknown mouse event type=4, code=5, value=0
@4000000064f227053513af44 unknown mouse event type=4, code=5, value=0
@4000000064f22705351462f4 unknown mouse event type=4, code=5, value=0
@4000000064f227053515ca54 unknown mouse event type=4, code=5, value=0
@4000000064f227070601394c unknown mouse event type=4, code=5, value=0
@4000000064f2270707490744 unknown mouse event type=4, code=5, value=0
@4000000064f22707074ad04c unknown mouse event type=4, code=5, value=0
@4000000064f2270707d8fad4 unknown mouse event type=4, code=5, value=0
@4000000064f2270708808d44 unknown mouse event type=4, code=5, value=0
@4000000064f2270709188784 unknown mouse event type=4, code=5, value=0
@4000000064f2270709ceeed4 unknown mouse event type=4, code=5, value=0
@4000000064f227070a76bbdc unknown mouse event type=4, code=5, value=0
@4000000064f227070b0f525c unknown mouse event type=4, code=5, value=0
@4000000064f227070bb7e2b4 unknown mouse event type=4, code=5, value=0
@4000000064f227070c506994 unknown mouse event type=4, code=5, value=0
@4000000064f227070ce8d51c unknown mouse event type=4, code=5, value=0
@4000000064f227070d9086cc unknown mouse event type=4, code=5, value=0
@4000000064f227070e2884f4 unknown mouse event type=4, code=5, value=0
@4000000064f227070ec12b14 unknown mouse event type=4, code=5, value=0
@4000000064f227070f598ae4 unknown mouse event type=4, code=5, value=0
@4000000064f227070ff1cf5c unknown mouse event type=4, code=5, value=0
@4000000064f2270710263954 unknown mouse event type=4, code=5, value=0
@4000000064f227090d188bf4 unknown mouse event type=4, code=5, value=0
@4000000064f227090e2711dc unknown mouse event type=4, code=5, value=0
[... more of the same...]
@4000000064f2271a0dcd41ac unknown mouse event type=4, code=5, value=0
@4000000064f2272b2ea55284 *** starting gui ***
@4000000064f2272b3132a484 *** headless device=0
@4000000064f2272b3137635c *** Waiting for localsettings...
@4000000064f2272b32294924 *** Localsettings is up, continuing...
@4000000064f2272b37528f3c *** Starting gui, with VNC enabled (VncLocal=1 VncInternet=1)
@4000000064f2272b3a5046ac QVNCServer created on port 5900
@4000000064f2272b3a8e9f9c Reloading input devices: "LinuxInput:/dev/input/event0"
@4000000064f2272c02c5117c "using /etc/venus for runtime features"
@4000000064f2272c02cb33cc "running on raspberrypi2"
@4000000064f2272c037959d4 Connecting to deprecated signal QDBusConnectionInterface::serviceOwnerChanged(QString,QString,QString)
@4000000064f2272c0ea8d0b4 Creating settings
@4000000064f2272c1b50ed74 [VeQItemDbusPublisher] Registered service "debug.victronenergy.gui"
@4000000064f2272c1b52e944 language is "en"
@4000000064f2272c1caec40c accessLevelChanged:  2
@4000000064f2272c1caefabc setAccessLevel:  2
@4000000064f2272c1d441a84 LocalThemeDaemonClient: Looking for assets in "/opt/victronenergy/themes/ccgx"
@4000000064f2272c1d4a2564 Screen: 800x480 282x169mm 72x72dpi
@4000000064f2272c2526b054 [Qwacs::qwacsRegisterd]  "com.victronenergy.qwacs"
@4000000064f2272c279c404c TimeZone::valueChanged:  "America/New_York"
@4000000064f2272c27a15514 ScreenSaver::dbusDisplayOffTimeChanged:  60
@4000000064f2272c27a2bc74 ScreenSaver::enable
@4000000064f2272c27a2c82c ScreenSaver::screenOn
@4000000064f2272c27ab922c Buzzer::setBuzzer: false
@4000000064f2272f35bb7864 Get Items failed "com.victronenergy.temperature.RpiCpu06"
[... 8 lines cut ...]
@4000000064f2273a38802644 unknown mouse event type=4, code=5, value=0
@4000000064f2273a3934d044 unknown mouse event type=4, code=5, value=0
@4000000064f2273a39cbe7f4 unknown mouse event type=4, code=5, value=0
@4000000064f2273a3a642c6c unknown mouse event type=4, code=5, value=0
@4000000064f2273a3b0f06b4 unknown mouse event type=4, code=5, value=0
@4000000064f2273b000af6a4 unknown mouse event type=4, code=5, value=0
@4000000064f2273b00b19eb4 unknown mouse event type=4, code=5, value=0
@4000000064f2273b01488b6c unknown mouse event type=4, code=5, value=0
@4000000064f2273b01f050a4 unknown mouse event type=4, code=5, value=0
@4000000064f2273b0279e364 unknown mouse event type=4, code=5, value=0
@4000000064f2273b031204b4 unknown mouse event type=4, code=5, value=0
@4000000064f2273b03aa31bc unknown mouse event type=4, code=5, value=0
@4000000064f2273d0e59ce24 file:///opt/victronenergy/gui/qml/OverviewGeneratorEnhanced.qml:329:4: QML OverviewAcValuesEnhanced: Binding loop detected fo>
[... cut ...]
@4000000064f2273d2a99c4f4 unknown mouse event type=4, code=5, value=0
@4000000064f2273d2a9b5f1c unknown mouse event type=4, code=5, value=0
@4000000064f2273d2a9d7a2c unknown mouse event type=4, code=5, value=0
[... more of the same...]
@4000000064f22742092141e4 unknown mouse event type=4, code=5, value=0
@4000000064f2277d33edf7dc ScreenSaver::timer
@4000000064f2277d33ee0b64 ScreenSaver::screenOff
@4000000064f228ef00b19eb4 QVNCServer new incoming connection from "127.0.0.1"
@4000000064f228ef01154e3c QVNCServer disconnected from "127.0.0.1"

dersch81, 2023-09-01:
Having the same issue here. But only if i use "gui mods ac coupled" with "gui mods simple"

trooster, 2023-09-02:
Same here. I switched temporarily back to the 'victon stock' flow in the guimod settings.

kwindrem, 2023-09-02:
fixed in v8.01 available now

remember to set the branch/tag back to 'latest'

lfhoward, 2023-09-02:
Done! And it works.

Thank you, Kevin! You rock!

mmeffects, 2023-09-08:
Hi Kevin,

Seems white Screen comes back with GUI Mods 8.03 on Venus OS 3.10 ~20
(cant upgrade atm to lates Venus OS Release, as it breaks Modbus TCP Smart Meter Integration SDM 630)

Let me know if you need further details...

kwindrem, 2023-09-08:
Should be fixed now in v8.04.

Don't forget to set the GuiMods branch/tag back to latest

closed 2023-09-08 by kwindrem