kwindrem/GuiMods #189: Venusos 3.10~38 Guimod 8.04 not compatible?

pieterjanvz, 2023-09-09:
i get the error Incompatible in setuphelper after Upgrading venusos to version 3.10~38

[image]

pieterjanvz, 2023-09-09:
after a restart , and retry i get this message:

[image]

kwindrem, 2023-09-09:
fixed now in v8.05

closed 2023-09-09 by kwindrem

kwindrem, 2023-09-09:
If you are still having problems, wait a while -- maybe 30 minutes or more. It sometimes takes a while for updates to GitHub to propagate through the internet.

pieterjanvz, 2023-09-09:
THANKS , its fixed , you are the best.

[image]