kwindrem/GuiMods #190: Additional devices

Ribodo, 2023-09-10:
It would be great if it were possible to add others devices from the Cerbo device list. One is an additional BMV712 that I have connected, but no way of showing on GUImods.

kwindrem, 2023-09-10:
If the BMV/Shunt is for something other than battery, then you can probably show it. Set the BMV to DC Meter then pick a sub-type (e.g., Alternator). Then use the GuiMods DC Coupled flow overview and a new tile will appear for that device.

If the BMV is for an alternate battery, then I agree that the entire GX infrastructure doesn't properly support it. Adding additional battery monitors isn't a simple task. If it's part of the main battery bank, then its contribution really needs to be factored in at the system level. If it is not part of the main battery bank, information about how power is exchanged with the rest of the system is needed and beyond the scope of GuiMods.

Ribodo, 2023-09-10:
Thanks for coming back. Indeed the BMV 712 is a stand alone Victron battery monitor for a 2nd set of batteries. This is connected by VE direct to the Cerbo and shows in the device list only. Their plug in for the B&G Zeus multi-function display shows it very well.

I had assumed it was just something to be shown, like the Ruuvi tags or the tank monitors.

Well thanks for taking the time to reply. Love the GUImods and will keep using it 😀

Cheers
Phil

iNotMario, 2023-10-09:
Unfortunately, victron made a pretty limited architecture. Multiple banks, and other complex systems cannot be represented by Victrons rigid structure.

I have a secondary bank that is Lead-Acid based for my alternator. I use a smart-shunt to monitor the current in-and-out from the combination of the batteries and alternator - so i have it show up as an Alternator in GUIMods. Can't monitor the battery state, but at least i can see the current flow for it.

buddapps, 2023-10-20:
Any possibility in the large image format to show the Dashboard for NodeRed?

It would also be nice to be able to slide in views of web pages for other devices on the local LAN...

kwindrem, 2023-10-20:
I think the NodeRed functionality in Venus OS is limited to server side stuff (no display). So probably not.

There is no web browser in Venus OS, plus it would be impossible to integrate that into QML, at least for someone at my coding level. So, again not possible.

closed 2023-10-20 by kwindrem